ihrapsa/OctoWrt #3: Unable to install, error during  pip install Octoprint==1.6.1 step

thomsonOZ, 2021-09-01:
Building wheel for regex (setup.py) ... error
  ERROR: Failed building wheel for regex
  Running setup.py clean for regex
Successfully built blinker emoji future pylru PyYAML
Failed to build regex
Installing collected packages: regex, PyYAML, pylru, pkginfo, OctoPrint-PiSupport, OctoPrint-FirmwareCheck, OctoPrint-FileCheck, netaddr, markdown, immutabledict, future, Flask-Login, Flask-Babel, Flask-Assets, filetype, feedparser, emoji, cachelib, blinker, Octoprint
    Running setup.py install for regex ... error
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-1swi3pu8/regex_6192760ef2914e4b97253488e9beffd6/setup.py'"'"'; __file__='"'"'/tmp/pip-install-1swi3pu8/regex_6192760ef2914e4b97253488e9beffd6/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-r8jlb22q/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/regex Check the logs for full command output.

ihrapsa, 2021-09-01:
Did you follow the steps carefully? That regex package needs to be downloaded from my repo and installed. Follow the Python3 steps then proceed to installing Octoprint with pip commnand

[image]

thomsonOZ, 2021-09-01:
My box currently is Klipper Fliudd ready box, do I need to reset it for OctoPrint?

root@KlipperWrt:/tmp# wget https://github.com/ihrapsa/OctoWrt/raw/main/packages/python3-regex_20
21-8-3_mipsel_24kc.ipk
pkg install *.ipk--2021-09-02 08:45:17--  https://github.com/ihrapsa/OctoWrt/raw/main/packages/python3-regex_2021-8-3_mipsel_24kc.ipk
Resolving github.com... 13.236.229.21
Connecting to github.com|13.236.229.21|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ihrapsa/OctoWrt/main/packages/python3-regex_2021-8-3_mipsel_24kc.ipk [following]
--2021-09-02 08:45:18--  https://raw.githubusercontent.com/ihrapsa/OctoWrt/main/packages/python3-regex_2021-8-3_mipsel_24kc.ipk
Resolving raw.githubusercontent.com... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com|185.199.108.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 230286 (225K) [application/octet-stream]
Saving to: 'python3-regex_2021-8-3_mipsel_24kc.ipk'

python3-regex_2021-8-3_ 100%[===============================>] 224.89K   406KB/s    in 0.6s

2021-09-02 08:45:19 (406 KB/s) - 'python3-regex_2021-8-3_mipsel_24kc.ipk' saved [230286/230286]

root@KlipperWrt:/tmp# opkg install *.ipk
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package python3-packages (1.0-1) installed in root is up to date.

ihrapsa, 2021-09-01:
You can have both if you disable klipper, moonraker and nginx services but I'd recommend starting fresh.

thomsonOZ, 2021-09-02:
I did a fresh install and it seem no error in installation.
Thanks!

thomsonOZ, 2021-09-02:
I did a fresh install and follow all steps, did not see any red error. Nothing shows in /mnt folder and unable to start OctoPrint service (with or without Printer connect)

ihrapsa, 2021-09-02:
> I did a fresh install and follow all steps, did not see any red error. Nothing shows in /mnt folder and unable to start OctoPrint service (with or without Printer connect)

What are you looking after in `/mnt`? Octoprint should automatically start on port `5000` just be patient as it takes around 5-7 min to first bootup.