ihrapsa/OctoWrt #4: error when installing octoprint on creality wifibox

Johan2356, 2021-11-02:
Hello, 
I follow all the steps mentioned in this tutorial, and receive an error while installing 
pip install Octoprint==1.6.1
errorcode:
 ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-lioir7bb/regex_8a13daca66ce4925aeb61af2b6081113/setup.py'"'"'; __file__='"'"'/tmp/pip-install-lioir7bb/regex_8a13daca66ce4925aeb61af2b6081113/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-go5593gk/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/regex Check the logs for full command output.

how can i solve this, all ideas are welcome.
i 'm a beginner with linux...

thanks in advance
Johan

ihrapsa, 2021-11-02:
You need to install the precompiled regex package before installing
octoprint. Did you do that step that's specified in the guide?

Jonah

Johan2356, 2021-11-02:
Hello, 
I started with installing mainsail, from a micro sd, then connected to wifi, so i can acces it , after that the steps below, until i had this error.
the file you mention here does not ring a bell with me...
[image]
[image]

ihrapsa, 2021-11-02:
I don't understand, you want Mainsail or Octoprint? Not recommend to have
them both. especially if your new to linux.
Regex is the cross compiled python3 package. Look at the last line from the
screenshos you've sent.

Johan2356, 2021-11-02:
I understand the confusion, 
First i had choosen mainsail, until I understood that the printer himself needed to be flashed too. 
Too risky for me,when that fails, i have nothing anymore,  so i continued installing octoprint, without erasing mainsail.
i think thats the problem.

ihrapsa, 2021-11-02:
No, that shouldn't cause a problem. However, it's recommended to start
fresh with either mainsail/fluidd or Octoprint so reset the box (hold reset
button ~7sec then proceed with the Octoprint steps. Note that after
resetting the box will still be on openwrt but youllt have to redo the
network settings)

You can't fail by flashing the mainbaord. your fear is a bit unjustified.
what mainboard do you have?

Do you know anything about klipper? if you're a beginner in this hobby and
don't want to try it you can just stick to marlin and get octoprint on the
box. It'll be laggy (booting up) but usable.

Johan2356, 2021-11-02:
I have a cr10s printer, 
Many thanks for this information, i will try this evening after work..

ihrapsa, 2021-11-02:
Just note that the printer model doesn't matter. Mainboard model matters
when flashing klipper. Though if you'll never flash it that doesn't matter
anymore.

Good luck!

closed 2021-11-15 by ihrapsa