ihrapsa/OctoWrt #6: error when installing octoprint 1.6.1

drakengard780, 2022-01-15:
hi, i have a problem , when i arrive to install octoprint compare an error:

ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jqid1cfe/regex_e2f3b767df6b4c37b68b350b894d6d2e/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jqid1cfe/regex_e2f3b767df6b4c37b68b350b894d6d2e/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-zpl7kz4f/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/regex Check the logs for full command output.

what can i do ? a read all step in wiki and video tutorial .

ihrapsa, 2022-01-15:
You sure you followed these steps?
[image]

drakengard780, 2022-01-15:
absolutely

drakengard780, 2022-01-15:
can i retry?

drakengard780, 2022-01-15:
[https://ibb.co/5MrxbBN](url)

drakengard780, 2022-01-15:
[image]

drakengard780, 2022-01-15:
if i reinstall python dependence compare thise warning 
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

ihrapsa, 2022-01-15:
> absolutely

not the octoprint command just the commands I screenshot above.

ihrapsa, 2022-01-15:
> if i reinstall python dependence compare thise warning
> WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

ignore that for the time being

drakengard780, 2022-01-15:
ok i insert second command all ok
i try to reinstall octoprint

drakengard780, 2022-01-15:
ok now install it but if go to ip:5000 stock to load same dead

drakengard780, 2022-01-15:
ok now all work ...to start up to 10 minute. but now now all work eccept path of webcam , if test it say path not exist

ihrapsa, 2022-01-15:
> ok now all work ...to start up to 10 minute. but now now all work eccept path of webcam , if test it say path not exist

what webcam are you using? is it plugged? what's the path set in octoprint?

drakengard780, 2022-01-15:
i not have for now webcam arrive tommorrow. but i set the path and install all the code in wiki guide but i not have the path

ihrapsa, 2022-01-15:
> i not have for now webcam arrive tommorrow. but i set the path and install all the code in wiki guide but i not have the path

there's no webcam path if you don't have webcam. what's the path that you set? show me

drakengard780, 2022-01-16:
Ok thank you this issue resolve

closed 2022-01-16 by drakengard780