ihrapsa/OctoWrt #8: mjpg-streamer config file settings

RobertHolscher, 2022-01-16:
First of all, excellent work on the tutorial. It works fine, except for the USB camera with mjpg-streamer. I don't get an image, so I tried to modify the /etc/config/mjpg-streamer
See file content below:
config mjpg-streamer 'core'
        option enabled '1'
        option input 'uvc'
        option output 'http'
        option device '/dev/video0'
        option resolution '1280x720'
        option yuv '0'
        option quality '80'
        option fps '10'
        option led 'auto'
        option www '/www/webcam'
        option port '8080'
        #option listen ip '192.168.1.1'
        #option username 'openwrt'
        #option password 'openwrt'

The option Enabled was default to 0 and I changed it to 1 and I also disabled the login for the stream. 
This was all done because I am not getting any image .
Could you maybe provide me with how this config should be ? Enabled '0' or '1' and is the device automatically changed to /dev/video0 ? and are there more options to check ?

ihrapsa, 2022-01-16:
Are you still not getting an image? that 'enabled' option did nothing for me so I kept it on 0. Did not read the docs of mjpg-stereamer config file.

RobertHolscher, 2022-01-16:
Hi Iona,

 

The problem is in the resolution . I have a camera with definition of 720p but with that resolution I am not getting an image. With playing around I found out that the yuv option had to be set to 1 for my camera to stream and the FPS has to be set to 10 or 20

Then I am getting finally an image. The enabled flag is now at 1 but I don’t want to mess around with too many parameters at the time

 

Kind regards

 

Robert

 

Deze e-mail is gecontroleerd op virussen door AVG.
http://www.avg.com

ihrapsa, 2022-01-16:
what's the output of `v4l2-ctl --all`? you should get an image on higher resolutions as well. How is the cpu usage with the yuv option to 1 while streaming?

RobertHolscher, 2022-01-17:
Dear Iona

 

Hereby the info . I don’t see much wrong with it , do you  ? 

It is a bit frustrating, since I sometimes have changed a setting in the mjpg-streamer config and then I see a picture, but the is seems to break and a restart of de mjpg-streamer demon does not bring the picture back

Any advice ?

 

 

 

***@***.***:~# v4l2-ctl --all

Driver Info:

        Driver name      : uvcvideo

        Card type        : USB 2.0 Camera: USB Camera

        Bus info         : usb-101c0000.ehci-1.4

        Driver version   : 5.10.46

        Capabilities     : 0x84a00001

                Video Capture

                Metadata Capture

                Streaming

                Extended Pix Format

                Device Capabilities

        Device Caps      : 0x04200001

                Video Capture

                Streaming

                Extended Pix Format

Priority: 2

Video input : 0 (Camera 1: ok)

Format Video Capture:

        Width/Height      : 640/480

        Pixel Format      : 'YUYV' (YUYV 4:2:2)

        Field             : None

       Bytes per Line    : 1280

        Size Image        : 614400

        Colorspace        : sRGB

        Transfer Function : Rec. 709

        YCbCr/HSV Encoding: ITU-R 601

        Quantization      : Default (maps to Limited Range)

        Flags             : 

Crop Capability Video Capture:

        Bounds      : Left 0, Top 0, Width 640, Height 480

        Default     : Left 0, Top 0, Width 640, Height 480

        Pixel Aspect: 1/1

Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags: 

Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 

Streaming Parameters Video Capture:

        Capabilities     : timeperframe

        Frames per second: 5.000 (5/1)

        Read buffers     : 0

                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0

                       contrast 0x00980901 (int)    : min=0 max=64 step=1 default=32 value=32

                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=64 value=64

                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0

white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1

                          gamma 0x00980910 (int)    : min=72 max=500 step=1 default=100 value=100

                           gain 0x00980913 (int)    : min=0 max=100 step=1 default=0 value=0

           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2

                                0: Disabled

                                1: 50 Hz

                                2: 60 Hz

      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive

                      sharpness 0x0098091b (int)    : min=0 max=6 step=1 default=3 value=3

         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1

                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3

                                1: Manual Mode

                                3: Aperture Priority Mode

              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=157 value=157 flags=inactive

         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1

***@***.***:~#

 

http://www.avg.com

ihrapsa, 2022-01-17:
Hi, Imy afraid your webcam is not mjpg. It's pixel format is yuyv. Could you try this command? `ls /dev/video*` then `v4l2-ctl --list-formats` and show me.

RobertHolscher, 2022-01-17:
Hi , I think the YUV is triggered by the mjpg-streamer config YUV = ‘1’

I have put this to zero again and now the output for v4l2-ctl –all is

 

Driver Info:

        Driver name      : uvcvideo

        Card type        : USB 2.0 Camera: USB Camera

        Bus info         : usb-101c0000.ehci-1.2

        Driver version   : 5.10.46

        Capabilities     : 0x84a00001

                Video Capture

                Metadata Capture

                Streaming

                Extended Pix Format

                Device Capabilities

        Device Caps      : 0x04200001

                Video Capture

                Streaming

                Extended Pix Format

Priority: 2

Video input : 0 (Camera 1: ok)

Format Video Capture:

        Width/Height      : 640/480

        Pixel Format      : 'MJPG' (Motion-JPEG)

        Field             : None

        Bytes per Line    : 0

        Size Image        : 614989

        Colorspace        : Default

        Transfer Function : Default (maps to Rec. 709)

        YCbCr/HSV Encoding: Default (maps to ITU-R 601)

        Quantization      : Default (maps to Full Range)

        Flags             : 

Crop Capability Video Capture:

        Bounds      : Left 0, Top 0, Width 640, Height 480

        Default     : Left 0, Top 0, Width 640, Height 480

        Pixel Aspect: 1/1

Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags: 

Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags: 

Streaming Parameters Video Capture:

        Capabilities     : timeperframe

        Frames per second: 5.000 (5/1)

        Read buffers     : 0

                     brightness 0x00980900 (int)    : min=-64 max=64 step=1 default=0 value=0

                       contrast 0x00980901 (int)    : min=0 max=64 step=1 default=32 value=32

                     saturation 0x00980902 (int)    : min=0 max=128 step=1 default=64 value=64

                            hue 0x00980903 (int)    : min=-40 max=40 step=1 default=0 value=0

white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1

                          gamma 0x00980910 (int)    : min=72 max=500 step=1 default=100 value=100

                           gain 0x00980913 (int)    : min=0 max=100 step=1 default=0 value=0

           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2

                                0: Disabled

                                1: 50 Hz

                                2: 60 Hz

      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive

                      sharpness 0x0098091b (int)    : min=0 max=6 step=1 default=3 value=3

         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1

                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3

                                1: Manual Mode

                                3: Aperture Priority Mode

              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=157 value=157 flags=inactive

         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1

***@***.***:/dev#

 

In /dev the command v4l2-ctl --list-formats gives

 

***@***.***:/dev# v4l2-ctl --list-formats

ioctl: VIDIOC_ENUM_FMT

        Type: Video Capture

 

        [0]: 'MJPG' (Motion-JPEG, compressed)

        [1]: 'YUYV' (YUYV 4:2:2)

 

My mjpg-streamer config is

config mjpg-streamer 'core'

        option enabled '0'

        option input 'uvc'

        option output 'http'

        option device '/dev/video0'

        option resolution '640x480'

        option yuv '0'

        option quality '80'

        option fps '5'

        option led 'auto'

        option www '/www/webcam'

        option port '8080'

        #option listen ip '192.168.1.1'

        #option username 'openwrt'

        #option password 'openwrt'

~

~

~

~

~

~

~

~

~

~

~

~

~

~

~

- /etc/config/mjpg-streamer 1/16 6%

 

http://www.avg.com

ihrapsa, 2022-01-17:
so your webcam has 2 video inputs. video0 with mjpg pixel format and video1 with yuyv.

Maybe mjpg can output a max resolution of 640x480.

let's try this command `v4l2-ctl --list-formats-ext`
This should output all resolutions and fps that each input can handle

RobertHolscher, 2022-01-17:
Hi , hereby the results of 

 

v4l2-ctl --list-formats-ext

 

ioctl: VIDIOC_ENUM_FMT

        Type: Video Capture

 

        [0]: 'MJPG' (Motion-JPEG, compressed)

                Size: Discrete 1280x720

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 640x480

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 352x288

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 320x240

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 176x144

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 160x120

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

        [1]: 'YUYV' (YUYV 4:2:2)

                Size: Discrete 1280x720

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 640x480

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 352x288

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 320x240

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 176x144

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

                Size: Discrete 160x120

                        Interval: Discrete 0.033s (30.000 fps)

                        Interval: Discrete 0.040s (25.000 fps)

                        Interval: Discrete 0.050s (20.000 fps)

                        Interval: Discrete 0.067s (15.000 fps)

                        Interval: Discrete 0.100s (10.000 fps)

                        Interval: Discrete 0.200s (5.000 fps)

***@***.***:~#

 

http://www.avg.com

ihrapsa, 2022-01-17:
Apparently mjpg pixel format can handle 720p. could you try setting that resolution in mjpg-streamer config file then do `service mjpg-streamer restart`

ihrapsa, 2022-01-17:
alternatively try forcing the resolution for video0: `v4l2-ctl --device /dev/video0 --set-fmt-video=width=1280,height=720`

RobertHolscher, 2022-01-17:
Hi , unfortunately no luck with both . Config was changed to 1280x720 and services restarted even with v4l2 directly the resolutions set but still no image by mjpg-streamer.

 

Any more ideas maybe ?

If I plug the web cam to my computer , it instantly works, so hardware seems in order

 

Kind regards

Robert

 

http://www.avg.com

RobertHolscher, 2022-01-17:
Hi , there is a mjpg-streamer Select() time out in the system log all the time 5 sec after starting the daemon.

Maybe this info helps

 

Mon Jan 17 12:04:08 2022 daemon.info octoprint[1934]: 2022-01-17 12:04:08,720 - octoprint.server.util.sockjs - INFO - User Holscher3D logged in on the socket from client 192.168.2.34

Mon Jan 17 12:06:35 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:06:36 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:06:52 2022 kern.info kernel: [ 6845.677534] usb 1-1.2: new high-speed USB device number 3 using ehci-platform

Mon Jan 17 12:06:52 2022 kern.info kernel: [ 6845.974716] uvcvideo: Found UVC 1.00 device USB 2.0 Camera (0c45:6366)

Mon Jan 17 12:06:52 2022 kern.info kernel: [ 6846.098426] input: USB 2.0 Camera: USB Camera as /devices/platform/101c0000.ehci/usb1/1-1/1-1.2/1-1.2:1.0/input/input0

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: starting application

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: MJPG Streamer Version.: 2.0

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: Using V4L2 device.: /dev/video0

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: Desired Resolution: 640 x 480

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: Frames Per Second.: 5

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: Format............: JPEG

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: TV-Norm...........: DEFAULT

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: www-folder-path......: /www/webcam/

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: HTTP TCP port........: 8080

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: HTTP Listen Address..: (null)

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: username:password....: disabled

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: commands.............: enabled

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: starting input plugin input_uvc.so

Mon Jan 17 12:06:54 2022 user.info MJPG-streamer [3309]: starting output plugin: output_http.so (ID: 00)

Mon Jan 17 12:07:14 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:07:14 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:07:55 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:07:55 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:08:36 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:08:37 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:09:17 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:09:18 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:09:59 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:09:59 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:10:40 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:10:40 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

Mon Jan 17 12:10:52 2022 user.info MJPG-streamer [3309]: select() timeout

Mon Jan 17 12:10:52 2022 user.info MJPG-streamer [3309]: cleaning up resources allocated by input thread

Mon Jan 17 12:11:21 2022 daemon.warn dnsmasq-dhcp[2579]: no address range available for DHCP request via wlan0

 

http://www.avg.com

ihrapsa, 2022-01-17:
this is quite weird. any chance to test it on another linux machine? I had similar issues with the creality webcam though I eventually found out that the mjpg pixel format was disabled in the webcam's firmware. Maybe the webcam can't really output 720p and it's just a marketing trick. It's just a guess though. What webcam/brand is that?

RobertHolscher, 2022-01-20:
> Hi
> I will check with an raspberry pi zero upcoming days
> And another web cam which I can borrow
>
> Will come back on this
> Thanks
> Robert
>
> Verstuurd vanaf mijn iPhone

Hi, 
I have stopped investigating the problem and am using an ESP32 cam now with an led light. This is an wifi solution with the right firmware uploaded to create 2 channels. One for the stream and one for the snapshot.
This works well

Thanks for your suggestions.
One other question: the update to version 1.7 octoprint fails all the time. Is this known?

Kind regard

Robert

Verstuurd vanaf mijn iPad

closed 2022-01-20 by RobertHolscher

ihrapsa, 2022-01-20:
Glad you found an alternative. Still curious what's happening with your
webcam. Since you mention esp32-cam, checkout esp3D project by luc on
GitHub. you can use the esp32-cam to control and monitor the printer. it's
an awesome project, lightweigt and it works with Marlin.

The guide tells you not toupdate (or at least it should) as any untested
update will most probably fail. Official Octoprint update system uses
Debian syntax/scripts or requiers packages that might need to be cross
compileded on openwrt. I'll try to test the latest version in my sparetime,
devs said it should be a bit more efficient.

Iona