ihrapsa/OctoWrt #9: Update octoprint to v1.7 fails

RobertHolscher, 2022-01-20:
I get notified that a new version of octoprint is available (1.7)
I do this through the software update feature in the octoprint web ui
It looks like everthing is going fine but in the end it fails.
Tomorrow i will add the log shown. 
i have tried it 3 times. One time with all my plugins active, ones with some plugins switched off and ones in safe mode.

ihrapsa, 2022-01-20:
yeah that's the official update system. I'd rather bypass it but any error log would help

TheFr0st3r, 2022-01-28:
so can I update it in some way?

Sebazzz, 2022-02-08:
> yeah that's the official update system. I'd rather bypass it but any error log would help

The core of the issue appears to be that it attempts to install netifaces 0.11.0 (to upgrade the existing 0.10.9), but it fails:

```
ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-mp07u60z/netifaces_abdd3d79aa2c4843b17901c0571602ee/setup.py'"'"'; __file__='"'"'/tmp/pip-install-mp07u60z/netifaces_abdd3d79aa2c4843b17901c0571602ee/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-0oqqb1_9/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/netifaces Check the logs for full command output.
```

Further investigation by installation with the `--verbose` flag reveals: 

[image]

Turns out this [option gcc does not recognize was introduced in gcc 8.0](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=70268), and the OpenWrt install runs gcc 7.9 with no option to upgrade.

Sebazzz, 2022-02-08:
Octoprint doesn't use a simple requirements.txt, and pip does not allow ignoring a dependency nor can you resolve dependency conflicts if you force the current netifaces version.

My thinking was to _fool_ pip that netifaces 0.11.0 is actually installed, but I can't find any information about where pip keeps the registry of installed packages.

ihrapsa, 2022-02-08:
Hi yeah I know it fails.  And you're right openwrt still uses an outdate
compiler. Also, octoprint has removed requirements.txt quite a while ago.
dependencies can be seen inside setup.py. Was planning to update the guide
for the latest version these days but if your interested I can help you
out. Usually when  pip fails due to compiler issues I look wheter there's a
python precompiled package into the opkg repository. (there is one but it's
0.10.9 still) If not I build it myself which is fairly straightforward.

Sebazzz, 2022-02-09:
Trying to cross-compile a package. I ran into:

```
In file included from /usr/include/signal.h:328,
                 from ./signal.h:52,
                 from c-stack.c:49:
c-stack.c:55:26: error: missing binary operator before token "("
   55 | #elif HAVE_LIBSIGSEGV && SIGSTKSZ < 16384
      |                          ^~~~~~~~
make[6]: *** [Makefile:1910: c-stack.o] Error 1
```

which led me to [this](https://www.mail-archive.com/search?l=openwrt-devel%40lists.openwrt.org&q=subject:%22Ubuntu+21.10+fails+to+compile+OpenWrt+21.02+hosts+tools+%5C%28m4%5C%29%22&o=newest&f=1) which led me to [this](https://github.com/openwrt/openwrt/pull/4675). Having manually applied the changes from that PR, got me compiling.

For anyone following along, [these docs are incredibly useful](https://openwrt.org/docs/guide-developer/toolchain/single.package). This works fine on WSL Ubuntu. 

Then, after I had installed the tools and toolchain, I modified the makefile in `package/feeds/packages/python3-netifaces` with the new version number and sha256 hash from https://files.pythonhosted.org/packages/source/n/netifaces/netifaces-0.11.0.tar.gz (which is 043a79146eb2907edf439899f262b3dfe41717d34124298ed281139a8b93ca32). 

Then it was just a matter of compiling `make package/python3-netifaces/compile` and in `/tmp/openwrt/bin/packages/mipsel_24kc/packages` the `python3-netifaces_0.11.0-1_mipsel_24kc.ipk` package was waiting for me.

@ihrapsa Feel free to include in the repository: [python3-netifaces_0.11.0-1_mipsel_24kc.ipk.zip](https://github.com/ihrapsa/OctoWrt/files/8036126/python3-netifaces_0.11.0-1_mipsel_24kc.ipk.zip) - I accidently compiled it against python 3.7 and due to [this issue](https://bugs.python.org/issue44035) the Python 3.9 build was broken so you need to symlink the python 3.7 library (`ln -fs /usr/lib/libpython3.9.so /usr/lib/libpython3.7.so.1.0`)

ihrapsa, 2022-02-09:
ooh I bow to you, sir! 😄 What I usually do is use the python3-packages manager/compiler in OpenWrt configuration (menuconfig) and that builds for me whichever python module version I want. 
[image]
Then Langauges -> Python
[image]
[image]
[image]

You can add multiple python modules and it will build a python3-packges.ipk containing what you've selected. Downside is that it will overwrite any previous python3-packges.ipk previously installed so I include everything that openwrt can't build (that's usually regex).

Did you manage to get 1.7.1 running?

Sebazzz, 2022-02-10:
Yes, Octoprint 1.7.**3** is now running fine!

RobertHolscher, 2022-02-10:
Wow! 
How can we as not so compiler saffy get it updated on our end?

Robert

Verstuurd vanaf mijn iPhone

closed 2022-10-15 by ihrapsa