ihrapsa/OctoWrt #10: Does OctoWrt work on Creality Box embedded in a CR-10s?

winnall, 2022-01-24:
I have a CR-10s, which has a Creality Box built into it. I assume that installing OctoWrt on it is no different from any other Creality Box. But will OctoWrt actually work with the CR-10s? Is there any other functionality important for the CR-10s in that Box?

Steve

ihrapsa, 2022-01-25:
Hi, OcoWrt will work on that box just like on the standalone box. the outside usb port connects to the box so you can plug a webcam there if needed. To flash openwrt on the box you can do it ota or the manual mode with the sd card though you need to upen the bottom plate of the printer

winnall, 2022-01-29:
Thank you. I'll try it out when I have time.

priyankchouksey, 2022-02-04:
I think you mean CR10 Smart. Yes, I just did a setup in the inside Wifi board. The only thing remaining to figure out is how to get the camera working. It is asking for credentials when the camera URL is accessed.

ihrapsa, 2022-02-04:
> I think you mean CR10 Smart. Yes, I just did a setup in the inside Wifi board. The only thing remaining to figure out is how to get the camera working. It is asking for credentials when the camera URL is accessed.

comment out the user and pass in `/etc/config/mjpg-streamer` file or add the user `openwrt` pass `openwrt` in the link

winnall, 2022-02-06:
> Hi, OcoWrt will work on that box just like on the standalone box. the outside usb port connects to the box so you can plug a webcam there if needed. To flash openwrt on the box you can do it ota or the manual mode with the sd card though you need to upen the bottom plate of the printer

I cannot flash the CR-10s. I've copied `cxsw_update.tar.bz2` (from ihrapsa/KlipperWrt) onto the root of the SD in the box, but the CR-10s ignores it and boots as usual.

[image]

ihrapsa, 2022-02-06:
> I cannot flash the CR-10s. I've copied `cxsw_update.tar.bz2` (from ihrapsa/KlipperWrt) onto the root of the SD in the box, but the CR-10s ignores it and boots as usual.
>
> [image]

Try this approache through it's web ui:

https://3dprintbeginner.com/how-to-install-klipperwrt-on-creality-box/

ihrapsa, 2022-02-08:
yeah, just follow the github guide... connect to that new AP and configure wifi or lan setup depending on whether you want it wired or wireless. then continue with the rest of the guide

winnall, 2022-02-10:
I got as far as

    pip install Octoprint==1.6.1

but the installation failed. So I tried the most recent version

    pip install Octoprint==1.7.3

and that failed with a similar error:

```
Using legacy 'setup.py install' for sarge, since package 'wheel' is not installed.
Using legacy 'setup.py install' for watchdog, since package 'wheel' is not installed.
Using legacy 'setup.py install' for blinker, since package 'wheel' is not installed.
Using legacy 'setup.py install' for emoji, since package 'wheel' is not installed.
Using legacy 'setup.py install' for future, since package 'wheel' is not installed.
Using legacy 'setup.py install' for markupsafe, since package 'wheel' is not installed.
Using legacy 'setup.py install' for netifaces, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pylru, since package 'wheel' is not installed.
Using legacy 'setup.py install' for wrapt, since package 'wheel' is not installed.
Using legacy 'setup.py install' for regex, since package 'wheel' is not installed.
Using legacy 'setup.py install' for pathtools, since package 'wheel' is not installed.
Installing collected packages: wrapt, sarge, regex, pytz, pylru, pkginfo, pathtools, OctoPrint-PiSupport, OctoPrint-FirmwareCheck, OctoPrint-FileCheck, netifaces, netaddr, ifaddr, filetype, emoji, colorlog, certifi, cachelib, blinker, zeroconf, werkzeug, websocket-client, watchdog, urllib3, semantic-version, PyYAML, pathvalidate, markupsafe, markdown, itsdangerous, immutabledict, idna, future, feedparser, Click, charset-normalizer, Babel, sentry-sdk, requests, Jinja2, flask, Flask-Login, Flask-Babel, Flask-Assets, Octoprint
  Running setup.py install for wrapt ... done
  Attempting uninstall: sarge
    Found existing installation: sarge 0.1.5.post0
    Uninstalling sarge-0.1.5.post0:
      Successfully uninstalled sarge-0.1.5.post0
  Running setup.py install for sarge ... done
  error: subprocess-exited-with-error
  
  × Running setup.py install for regex did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Running setup.py install for regex ... error
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> regex

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.
```

I found out that recent versions of regex are incompatible with OctoPrint (https://github.com/OctoPrint/OctoPrint/issues/4377), but I don't know enough about Python to know what to change and where.

Any thoughts?

ihrapsa, 2022-02-10:
Hi, that's for python 2 but in openwrt that pip command installs python 3 octoprint version since only python3 is installed. That regex error is most likely a version mismatch. There's another issue #9 about updating issues. I'm working these days on getting the latest version installed

ihrapsa, 2022-02-10:
> Finally got it up and running v1.7.3. Thanks for your help.
> 
> Somehow, during installation of Klipper, I got Python2 installed too. Perhaps it was when I was distracted by MainSail and tried to install that. However, once I removed all the python2 packages and installed your regex package and the netifaces package from #9, everything went perfectly.
> 
> Fast, it isn't :-) It reminds me of the late '80s. But I hope it will be adequate for starting a print a day.
> 
> Thanks for all the work in making this possible.

Yes, it's pretty slow on boot up.  make sure you don't have klipper/moonraker installed and running as well.  klipper with mainsail/fluidd would be a better alternative for this device but you lose the printer display.

winnall, 2022-02-15:
My printer is, in fact, a CR-10 Smart, which is not the same as a CR-10S, it seems, which you actually pointed out.

Since I’ve had OctoPrint installed on the printer, the initial layer is appallingly bad (nozzle too high). Is it possible that ABL is implemented on the internal Creality Box and that I have nuked that by installing OctoWrt?

ihrapsa, 2022-02-15:
Hi, ABL is setup in marlin and saved in eeprom afaik. Did you reflash marilin on the mainboard or factory reset it?

winnall, 2022-02-15:
I did a factory reset at one point.

ihrapsa, 2022-02-15:
then you've lost your abl settings. You need to recalibrate

winnall, 2022-02-15:
But I have done that several times since the reset using the option “Auto levelling” on the LED screen.

ihrapsa, 2022-02-15:
I don't understand. Your ABL calibration settings are not saved?

winnall, 2022-02-16:
I don’t know. All I know is that my print head is too far away from the bed with the corresponding messy first layer. Before I installed OctoWrt this wasn’t happening. Now it is. I’m not saying it necessarily has anything to do with OctoWrt, I’m just looking for causes.

I’ve got Creality Slicer on factory settings (which might be wrong, of course). But my suspicion, from a position of more or less complete ignorance, has been that ABL is not working for some reason.

ihrapsa, 2022-02-16:
As I said, octoprint has nothing to do with this. ABL is managed by marlin. AFAIK cr10 smart had issues with abl on release (it was not saving abl settings) but I think got fixed on a subsequent firmware release. Make sure you've got creality's latest custom marlin firmware for that, or switch altogether to proper marlin if there's any firmware out there

winnall, 2022-02-23:
This is resolved, and, as @ihrapsa said, it has nothing to do with OctoPrint. If anyone is interested, see https://3dprinting.stackexchange.com/questions/18932/how-do-you-make-a-new-cr-10-smart-do-auto-bed-levelling .

closed 2022-02-23 by winnall