ihrapsa/OctoWrt #14: OctoWrt Install error

Dahlaruski, 2022-02-13:
First, thank you for your work on this project.

I am fairly new to this, and followed your instructions provided to the letter. However, I constantly get the following error on the install:

8e16642c96758cd231d9595ab90912c91b282
  Stored in directory: /tmp/.cache/wheels/b7/a5/c4/504d913c2a55bb09c607541578ec5f844d1ff33467abe93ba5
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for regex (setup.py) ... error
  ERROR: Failed building wheel for regex
  Running setup.py clean for regex
Successfully built blinker emoji future pylru PyYAML
Failed to build regex
Installing collected packages: regex, pyserial, pylru, pkginfo, OctoPrint-PiSupport, OctoPrint-FirmwareCheck, OctoPrint-FileCheck, netaddr, filetype, emoji, colorlog, cachelib, blinker, PyYAML, pathvalidate, markdown, future, feedparser, Flask-Login, Flask-Babel, Flask-Assets, Octoprint
  error: subprocess-exited-with-error

  × Running setup.py install for regex did not run successfully.
  │ exit code: 1
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
  Running setup.py install for regex ... error
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> regex

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.

I am afraid I do not know enough about Python to figure out what is going wrong. It appears everything else installs correctly. Can you help?

ihrapsa, 2022-02-13:
Did you install the precompiled packages?
[image]

Dahlaruski, 2022-02-13:
Thanks for the quick reply. Yes I have. I attempted to rerun that command and I get:

root@OctoWrt:~# cd /tmp
root@OctoWrt:/tmp# wget https://github.com/ihrapsa/OctoWrt/raw/main/packages/python3.9-regex-2022.1.18+netifaces-0.11.0_mipsel_24kc.ipk
*.ipk--2022-02-13 15:41:45--  https://github.com/ihrapsa/OctoWrt/raw/main/packages/python3.9-regex-2022.1.18+netifaces-0.11.0_mipsel_24kc.ipk
Resolving github.com... 192.30.255.113
Connecting to github.com|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/ihrapsa/OctoWrt/main/packages/python3.9-regex-2022.1.18%2Bnetifaces-0.11.0_mipsel_24kc.ipk [following]
--2022-02-13 15:41:46--  https://raw.githubusercontent.com/ihrapsa/OctoWrt/main/packages/python3.9-regex-2022.1.18%2Bnetifaces-0.11.0_mipsel_24kc.ipk
Resolving raw.githubusercontent.com... 185.199.111.133, 185.199.110.133, 185.199.109.133, ...
Connecting to raw.githubusercontent.com|185.199.111.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 241522 (236K) [application/octet-stream]
Saving to: 'python3.9-regex-2022.1.18+netifaces-0.11.0_mipsel_24kc.ipk'

python3.9-regex-2022.1.18+net 100%[=================================================>] 235.86K  --.-KB/s    in 0.08s

2022-02-13 15:41:46 (2.93 MB/s) - 'python3.9-regex-2022.1.18+netifaces-0.11.0_mipsel_24kc.ipk' saved [241522/241522]

root@OctoWrt:/tmp# opkg install *.ipk
Installing python3-packages (1.0-1) to root...
Collected errors:
 * check_data_file_clashes: Package python3-packages wants to install file /usr/lib/python3.9/site-packages/netifaces-0.11.0-py3.9.egg-info/PKG-INFO
        But that file is already provided by package  * python3-netifaces
 * check_data_file_clashes: Package python3-packages wants to install file /usr/lib/python3.9/site-packages/netifaces-0.11.0-py3.9.egg-info/SOURCES.txt
        But that file is already provided by package  * python3-netifaces
 * check_data_file_clashes: Package python3-packages wants to install file /usr/lib/python3.9/site-packages/netifaces-0.11.0-py3.9.egg-info/dependency_links.txt
        But that file is already provided by package  * python3-netifaces
 * check_data_file_clashes: Package python3-packages wants to install file /usr/lib/python3.9/site-packages/netifaces-0.11.0-py3.9.egg-info/top_level.txt
        But that file is already provided by package  * python3-netifaces
 * check_data_file_clashes: Package python3-packages wants to install file /usr/lib/python3.9/site-packages/netifaces-0.11.0-py3.9.egg-info/zip-safe
        But that file is already provided by package  * python3-netifaces
 * check_data_file_clashes: Package python3-packages wants to install file /usr/lib/python3.9/site-packages/netifaces.cpython-39.so
        But that file is already provided by package  * python3-netifaces
 * opkg_install_cmd: Cannot install package python3-packages.

ihrapsa, 2022-02-13:
do `opkg install *ipk --force-overwrite` with that package

Dahlaruski, 2022-02-13:
Looks like that did the trick,

Thank you so much!