ihrapsa/OctoWrt #15: Problem with Hue plugin install in octoprint

RobertHolscher, 2022-02-23:
In the octowrt on the creality wifi box, i dont get the Hue plugins installed. There are 3 plugins in the repository but all fail. What can be the cause? Is there a workaround ? Please look at the attached messages. I notice that it tries to work with Wheel, resource. Is that the problem?
[image]
[image]

RobertHolscher, 2022-02-23:
Hereby the list of the tried plugins
[image]

RobertHolscher, 2022-02-23:
Hereby the logfile
[plugin_pluginmanager_console.log](https://github.com/ihrapsa/OctoWrt/files/8125742/plugin_pluginmanager_console.log)

ihrapsa, 2022-02-23:
the issue is with the numpy module. I don't htink mips architecture is able to build that package from source. I don't see a precompiled official package either. Only way to install that would be if we cross compile it on a different machine. I cantry crosscompiling it for you these days.

ihrapsa, 2022-02-23:
in the meantime try executing this `pip install --upgrade pip setuptools wheel` then try installing that plugin again

RobertHolscher, 2022-02-23:
Thanks I will try this

Verstuurd vanaf mijn iPhone

RobertHolscher, 2022-02-23:
Hi 
I tried it and it still fails to install the plugin, however not on the same place in the installer. Now it ends with the message that the plug-in fails to install, look at the log for details. I will also try the other hue plug-ins later

Kind regards 
Robert

Verstuurd vanaf mijn iPhone

ihrapsa, 2022-02-23:
can you share the logs?

RobertHolscher, 2022-02-23:
It is now sropping at download numpy
[plugin_pluginmanager_console 4.log](https://github.com/ihrapsa/OctoWrt/files/8128478/plugin_pluginmanager_console.4.log)

ihrapsa, 2022-02-23:
is it hanging there, no error?

RobertHolscher, 2022-02-24:
Correct

duchengyao, 2022-05-22:
Build numpy will cost long time. For about one hour in raspberry pi. In Ubuntu, I will install numpy using apt-get.

closed 2022-10-15 by ihrapsa