ihrapsa/OctoWrt #17: Octoprint unusable slow

mightymartin, 2022-03-10:
Hi,

dont know if its normal or the bounds of the hardware, but my octoprint needs nearly 10 minutes start. Is it only me, or normal on this box?

Octowrt itself is accessible after around 30 sec, which is fine. 

`2022-03-09 16:58:07,373 - octoprint.startup - INFO - ******************************************************************************
2022-03-09 16:58:07,420 - octoprint.startup - INFO - Starting OctoPrint 1.7.3
2022-03-09 16:58:07,455 - octoprint.startup - INFO - ******************************************************************************
2022-03-09 16:58:22,581 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity changed from offline to online
2022-03-09 16:58:22,683 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2022-03-09 16:58:22,700 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 1.1.1.1:53 is working
2022-03-09 16:58:22,712 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2022-03-09 16:58:25,021 - octoprint.startup - INFO - Blacklist processing done, adding 3 blacklisted plugin versions: roomtemp (any), GcodeEditor (>=0.1.1,<=0.2.8), gcodeleveling (>=0.1.0,<=0.1.1)
2022-03-09 16:59:01,616 - octoprint.plugin.core - INFO - Loading plugins from /usr/lib/python3.9/site-packages/octoprint/plugins, /.octoprint/plugins and installed plugin packages...
2022-03-09 16:59:40,073 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.3.1) did not pass check, not loading.
2022-03-09 16:59:46,376 - octoprint.plugin.core - INFO - Found 17 plugin(s) providing 17 mixin implementations, 34 hook handlers
2022-03-09 16:59:50,931 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2022-03-09 16:59:51,208 - octoprint.server - INFO - Intermediary server started
2022-03-09 16:59:52,297 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /.octoprint/uploads...
2022-03-09 16:59:52,329 - octoprint.filemanager.storage - INFO - ... file metadata for /.octoprint/uploads initialized successfully.
2022-03-09 16:59:52,440 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2022-03-09 16:59:52,465 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2022-03-09 16:59:52,482 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2022-03-09 16:59:52,521 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2022-03-09 16:59:52,544 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
2022-03-09 16:59:52,571 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2022-03-09 16:59:52,596 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2022-03-09 16:59:52,622 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2022-03-09 16:59:52,647 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2022-03-09 16:59:52,674 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2022-03-09 16:59:52,709 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_install'), Need(method='role', value='plugin_pluginmanager_manage')")
2022-03-09 16:59:52,735 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2022-03-09 16:59:52,759 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2022-03-09 16:59:52,789 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2022-03-09 17:00:32,838 - octoprint.util.pip - INFO - Using "/usr/bin/python3 -m pip" as command to invoke pip
2022-03-09 17:01:13,408 - octoprint.util.pip - INFO - Version of pip is 21.2.4
2022-03-09 17:01:13,444 - octoprint.util.pip - INFO - pip installs to /usr/lib/python3.9/site-packages (writable -> yes), --user flag needed -> no, virtual env -> no
2022-03-09 17:01:13,459 - octoprint.util.pip - INFO - ==> pip ok -> yes
2022-03-09 17:01:16,620 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
2022-03-09 17:01:20,050 - octoprint.plugin.core - INFO - Initialized 17 plugin implementation(s)
2022-03-09 17:01:20,400 - octoprint.plugin.core - INFO - 17 plugin(s) registered with the system:
|  Action Command Notification Support (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/backup
|  Core Wizard (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/errortracking
|  Event Manager (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/eventmanager
|  File Check (2021.2.23) (bundled) = /usr/lib/python3.9/site-packages/octoprint_file_check
|  Firmware Check (2021.10.11) (bundled) = /usr/lib/python3.9/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/gcodeviewer
|  Logging (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/logging
|  Plugin Manager (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/pluginmanager
|  Software Update (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/softwareupdate
|  Virtual Printer (bundled) = /usr/lib/python3.9/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2022-03-09 17:01:20,723 - octoprint.environment - INFO - Detected environment is Python 3.9.10 under Linux (linux). Details:
|  hardware:
|    cores: 1
|    freq: unknown
|    ram: 126427136
|  os:
|    bits: 32
|    id: linux
|    platform: linux
|  python:
[... 14 lines cut ...]
2022-03-09 17:01:55,625 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on OctoWrt._http._tcp.local.' for _http._tcp.local.
2022-03-09 17:01:57,702 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on OctoWrt._octoprint._tcp.local.' for _octoprint._tcp.local.
2022-03-09 17:01:57,801 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on OctoWrt for SSDP
2022-03-09 17:01:57,982 - octoprint.server - INFO - Listening on http://0.0.0.0:5000 and http://[::]:5000
2022-03-09 17:01:59,474 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
[... cut ...]
2022-03-09 17:04:53,054 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 3.2s
2022-03-09 17:06:12,044 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2022-03-09 17:06:13,731 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2022-03-09 17:07:49,523 - octoprint.server.util.flask - INFO - Passively logging in user admin from 192.168.33.128
2022-03-09 17:07:49,554 - octoprint.access.users - INFO - Logged in user: admin
`

ihrapsa, 2022-03-10:
Don't know if you've skimmed through the guide but it's  prey [normal](https://github.com/ihrapsa/OctoWrt#6-reboot-and-wait-a-while) to wait that long at bootup. Octoprint plugin system bloated it lately. If you don't care about plugins, octoprint versions before plugins were introduced are pretty snappy at bootup too.

mightymartin, 2022-03-12:
Hi ,thanks for the information. Shame on me, found the boottime in the guide. Will try an old octoprint version. I dont need plugins.

closed 2022-03-12 by mightymartin

andy75ita, 2022-03-17:
same for me .... after 30 minutes octoprint dont open on my web browser ....

ihrapsa, 2022-03-17:
> same for me .... after 30 minutes octoprint dont open on my web browser ....

Did you watch the logs? Is octoprint properly installed? Did you flash the Octowrt openwrt bin?

andy75ita, 2022-03-17:
yes, all installed correctly using the guide, but i dont know what happening, i opened a issue and i have attached the log ;-)

dopehopes, 2022-05-08:
same problem with a 15 minutes boot. where can i find older versions?

ihrapsa, 2022-05-08:
> same problem with a 15 minutes boot. where can i find older versions?

Version [1.1.2](https://github.com/OctoPrint/OctoPrint/releases/tag/1.1.2) seems to be the last one before plugins were intorduced. You need to install this manually with python2 beacause pip versions of octoprint started above [v1.3.11](https://pypi.org/project/OctoPrint/#history) so `pip install octoprint=v1.1.2` will not work.

To get it installed download the v1.1.2 source code -> `cd` into root directory -> do `python2 setup.py install` 

You'll probably get some errors so let me know