ihrapsa/OctoWrt #18: PIP Install - Error relocating /usr/lib/libpython3.10.so.1.0:

collaboratips, 2022-03-13:
Hello, 
Thanks for your help.

During the installation I get stuck after dependencies installation.
just after : opkg install v4l-utils.... the wget command starts to failed (Error relocating).
I removed wget-ssl and reinstalled wget and I could go on.

But after Python3 installation, trying to install Octoprint I got those errors : 

root@KlipperWrt:/# pip install Octoprint==1.7.3
Error relocating /usr/lib/libpython3.10.so.1.0: __getitimer_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __setitimer_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __sigtimedwait_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __pthread_cond_timedwait_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __dlsym_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __lstat_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __gettimeofday_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __stat_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __wait3_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __localtime64_r: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __utimensat_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __futimens_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __clock_gettime64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __fstatat_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __getrusage_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __sem_timedwait_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __clock_settime64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __fstat_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __clock_getres_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __select_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __mktime64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __wait4_time64: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __gmtime64_r: symbol not found
Error relocating /usr/lib/libpython3.10.so.1.0: __sched_rr_get_interval_time64: symbol not found
root@KlipperWrt:/# 

I'm lost.
I restarted the installation from scratch, but same thing.

Thank you for your help.

ihrapsa, 2022-03-13:
Why don't you follow the Automatic steps?

collaboratips, 2022-03-13:
Thank you for your answer.
I tried, same result.
At a moment I'm stuck too, same error, and the installation failed :( .
I blame my poor linux knowledges.

ihrapsa, 2022-03-13:
Are you using the new octowrt image linked?

collaboratips, 2022-03-13:
In fact, I started with the tuto of Hypermakes, the goal is to install of course Octoprint.

ihrapsa, 2022-03-13:
Read the github repo. Follow the Automatic steps

collaboratips, 2022-03-13:
I flashed the :  fluiddWrt-FACTORY-ramips-mt76x8-creality_wb-01-squashfs-k5.10.46.binn.bin from KlipperWrt v2.1.

Then after configuration and the first script execution , I switched to you (wonderful, thank you :) ), git.

collaboratips, 2022-03-13:
Ok I got it, I'm gonna flash your version and I keep you posted.
Thanl you !!!

ihrapsa, 2022-03-13:
Read the github repo guide. You need to flash the Octowrt image

collaboratips, 2022-03-13:
thank you and sorry because I didn't follow your steps ! 
Very nice too take the time to help.

collaboratips, 2022-03-13:
@ihrapsa, 
It works ! thank you very much again.
I just have to find a way to use my cam.

closed 2022-03-18 by ihrapsa