ihrapsa/OctoWrt #19: how many time to see octoprint started ??

andy75ita, 2022-03-17:
After 30 minutes octoprint im unable to access to octoprint via web. OpenWRT was ok after only 2/3 minutes, but octoprint not ... whats happen ? below the system log:

Wed Mar 16 17:28:02 2022 kern.notice kernel: [    0.000000] Linux version 5.4.171 (runner@fv-az41-617) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r16490-781b5d8f30)) #0 Sun Feb 13 17:28:05 2022
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Board has DDR2
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Analog PMU set to hw control
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Digital PMU set to hw control
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] MIPS: machine is Creality WB-01
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Mar 16 17:28:02 2022 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Wed Mar 16 17:28:02 2022 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Zone ranges:
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Early memory node ranges
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Wed Mar 16 17:28:02 2022 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Wed Mar 16 17:28:02 2022 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Wed Mar 16 17:28:02 2022 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Wed Mar 16 17:28:02 2022 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Wed Mar 16 17:28:02 2022 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Wed Mar 16 17:28:02 2022 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Wed Mar 16 17:28:02 2022 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Writing ErrCtl register=0005f6e9
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Readback ErrCtl register=0005f6e9
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] Memory: 122236K/131072K available (4797K kernel code, 201K rwdata, 1060K rodata, 1228K init, 197K bss, 8836K reserved, 0K cma-reserved)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] NR_IRQS: 256
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] intc: using register map from devicetree
Wed Mar 16 17:28:02 2022 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x358/0x54c with crng_init=0
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] CPU Clock: 580MHz
Wed Mar 16 17:28:02 2022 kern.crit kernel: [    0.000000] timer_probe: no matching timers found
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.000010] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.015391] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.059463] pid_max: default: 32768 minimum: 301
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.068801] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.083141] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.104870] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.124071] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.137581] pinctrl core: initialized pinctrl subsystem
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.148973] NET: Registered protocol family 16
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.194713] clocksource: Switched to clocksource MIPS
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.206074] NET: Registered protocol family 2
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.214849] IP idents hash table entries: 2048 (order: 2, 16384 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.229791] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.246271] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.261343] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.275269] TCP: Hash tables configured (established 1024 bind 1024)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.287974] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.300832] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.314903] NET: Registered protocol family 1
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.323444] PCI: CLS 0 bytes, default 32
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.335871] workingset: timestamp_bits=14 max_order=15 bucket_order=1
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.357170] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.368630] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.406602] mt7621_gpio 10000600.gpio: registering 32 gpios
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.417895] mt7621_gpio 10000600.gpio: registering 32 gpios
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.429122] mt7621_gpio 10000600.gpio: registering 32 gpios
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.440366] Serial: 8250/16550 driver, 3 ports, IRQ sharing disabled
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.454114] printk: console [ttyS0] disabled
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.462558] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 28, base_baud = 2500000) is a 16550A
Wed Mar 16 17:28:02 2022 kern.info kernel: [    0.480359] printk: console [ttyS0] enabled
[... 401 lines cut ...]
Thu Mar 17 10:32:16 2022 daemon.warn dnsmasq-dhcp[2654]: no address range available for DHCP request via wlan0
Thu Mar 17 10:32:57 2022 daemon.warn wpa_supplicant[1472]: wlan0: Undefined secondary channel: drop OBSS scan results
Thu Mar 17 10:35:08 2022 daemon.info octoprint[1996]: [33m2022-03-17 10:35:08,661 - zeroconf - WARNING - Error with socket 7 (('127.0.0.1', 5353))): [Errno 22] Invalid argument[0m
Thu Mar 17 10:35:08 2022 daemon.info octoprint[1996]: Traceback (most recent call last):
Thu Mar 17 10:35:08 2022 daemon.info octoprint[1996]:   File "/usr/lib/python3.9/asyncio/selector_events.py", line 1056, in sendto
Thu Mar 17 10:35:08 2022 daemon.info octoprint[1996]: OSError: [Errno 22] Invalid argument[0m
[... cut ...]
Thu Mar 17 10:53:14 2022 daemon.info octoprint[1996]: [37m2022-03-17 10:53:14,758 - octoprint.server.util.flask - INFO - Passively logging in user admin from 192.168.1.8[0m
Thu Mar 17 10:53:14 2022 daemon.info octoprint[1996]: [37m2022-03-17 10:53:14,764 - octoprint.access.users - INFO - Logged in user: admin[0m
Thu Mar 17 10:53:58 2022 daemon.warn wpa_supplicant[1472]: wlan0: Undefined secondary channel: drop OBSS scan results
Thu Mar 17 10:56:58 2022 daemon.warn wpa_supplicant[1472]: wlan0: Undefined secondary channel: drop OBSS scan results

ihrapsa, 2022-03-18:
What address are you using to access the Octoprint UI?

andy75ita, 2022-03-21:
hello :-) im using local ip (192.168.1.153:5000)

ihrapsa, 2022-03-21:
Did you change the lan static ip to a different subnet (other than 1)?

andy75ita, 2022-03-21:
no, im alwayse using 192.168.1.####

ihrapsa, 2022-03-21:
I don't understand. I think you're confusing the static lanip with the dhcp assigned ip. Static lan ip subnet of the box should be different  from 1. While dhcp ip subnet is always the same as your router's

ihrapsa, 2022-03-21:
Where is octofarm hosted?

> please consider i have a farm of 9 printers, i have this problem only with 2 box. Im using Octofarm, do you think Octofarm can be a problem ?

andy75ita, 2022-03-21:
sorry, you are right: ip was always 19.168.1.#### / same on my router 192.168.1.1 (did you mean this?)
Octofarm was hosted on Rasberry PI 3 model B

ihrapsa, 2022-03-21:
> sorry, you are right: ip was always 19.168.1.#### / same on my router 192.168.1.1 (did you mean this?)
> Octofarm was hosted on Rasberry PI 3 model B

the wifi ip of the BOX is 192.168.1.x because your router is assigning that ip. What I'm asking is if the lan static ip of the BOX is different than 192.168.1.1

andy75ita, 2022-03-21:
yes , the LAN ip is different (was 192.168.3.###)

ihrapsa, 2022-03-21:
have you tried starting from srcatch on the non working boxes?

andy75ita, 2022-03-22:
not tryed, but i can do it. Do you mean by resetting using the usb and the image file from creality ? then start from zero ?

ihrapsa, 2022-03-22:
Read the readme. No need to restore the stock image once you've alrt flashed Openwrt. There are a couple of ways to reset/start fresh. Check the readme

andy75ita, 2022-03-22:
ok i will check the readme file. Thanks for your time :-)

ihrapsa, 2022-03-22:
[Readme](https://github.com/ihrapsa/OctoWrt/tree/main/firmware/OpenWrt_snapshot#if-your-box-is-already-on-openwrt-and-has-the-luci-web-ui-reachable)

closed 2022-03-28 by ihrapsa

andy75ita, 2022-03-31:
ok reinstalled fron scratch, now octoprint was 100% operative after aprox 15 minutes from startup, so now i think is ok, thx for you help :-)

dopehopes, 2022-05-08:
> [Readme](https://github.com/ihrapsa/OctoWrt/tree/main/firmware/OpenWrt_snapshot#if-your-box-is-already-on-openwrt-and-has-the-luci-web-ui-reachable)

where can i find older versions for a faster boot?

ihrapsa, 2022-05-08:
> > [Readme](https://github.com/ihrapsa/OctoWrt/tree/main/firmware/OpenWrt_snapshot#if-your-box-is-already-on-openwrt-and-has-the-luci-web-ui-reachable)
> 
> where can i find older versions for a faster boot?

Version [1.1.2](https://github.com/OctoPrint/OctoPrint/releases/tag/1.1.2) seems to be the last one before plugins were intorduced. You need to install this manually with python2 beacause pip versions of octoprint started above [v1.3.11](https://pypi.org/project/OctoPrint/#history) so `pip install octoprint=v1.1.2` will not work.

To get it installed download the v1.1.2 source code -> `cd` into root directory -> do `python2 setup.py install` 

You'll probably get some errors so let me know