ihrapsa/OctoWrt #20: RPC call to luci/getLocaltime failed with error -32002: Access denied

doczong, 2022-05-05:
Hi All, 

After successfully installing KlipperWRT, I went over to this distro.  Installed same, no errors.

After reboot after Step #2 completes without error, getting the above when accessing System/System.

Also, OctoPrint isn't starting on Port 5000.

Otherwise, it seems to have installed fine.  df reports more than enough disk space free.

Any advice?

doczong, 2022-05-05:
Full Error:

RPCError

RPC call to luci/getLocaltime failed with error -32002: Access denied
  at handleCallReply (http://192.168.7.21:81/luci-static/resources/rpc.js?v=2.2.4:11:3)

ihrapsa, 2022-05-05:
Hi, did you install both on the same box? What step 2 are you refering to?

doczong, 2022-05-05:
Sorry, so this is two issues, the former re Step #2 is the Step #2 under Automatic on the main page of this repo:

cd /tmp
wget https://github.com/ihrapsa/OctoWrt/raw/main/scripts/2_octoprint_install.sh
chmod +x 2_octoprint_install.sh
./2_octoprint_install.sh

That said, OctoPrint does work.  It just took a really long time to boot.  I was able to connect it, and even connect Cura, so that is fine.  The only error is the "localtime failed" which I see someone has that exact issue on the current OpenWRT thread with Luci as well.  https://forum.openwrt.org/t/openwrt-21-02-2-second-service-release/121249/102

ihrapsa, 2022-05-05:
> Sorry, so this is two issues, the former re Step #2 is the Step #2 under Automatic on the main page of this repo:
> 
> cd /tmp wget https://github.com/ihrapsa/OctoWrt/raw/main/scripts/2_octoprint_install.sh chmod +x 2_octoprint_install.sh ./2_octoprint_install.sh
> 
> That said, OctoPrint does work. It just took a really long time to boot. I was able to connect it, and even connect Cura, so that is fine. The only error is the "localtime failed" which I see someone has that exact issue on the current OpenWRT thread with Luci as well. https://forum.openwrt.org/t/openwrt-21-02-2-second-service-release/121249/102
> 
> However, sometime after installing everything from Step #2, Luci stops working on the System Pages.

Oh I see. I think I've seen those errors from time to time but didn't look into them as it did not seem to cause any problems.

Yeah it takes a while to boot up because of the plugin manangement system. Did you by any chance installed it alongside klipper cause that's a no no?

doczong, 2022-05-06:
Perfect, just one more tidbit for your next build:
https://github.com/openwrt/luci/issues/5622

Looks like the solution is to update Luci as the version running on OctoWRT after updating all the OctoPrint packages is out of date.  

The use of getLocaltime has been dropped with https://github.com/openwrt/luci/commit/68815c88b0ebcfd0116070510a4c43ccff1d0310 - seems you're running an incoherent package selection. Make sure that all LuCI packages, in particular luci-mod-system are updated to the latest version.

ihrapsa, 2022-05-06:
> Perfect, just one more tidbit for your next build: [openwrt/luci#5622](https://github.com/openwrt/luci/issues/5622)
> 
> Looks like the solution is to update Luci as the version running on OctoWRT after updating all the OctoPrint packages is out of date.
> 
> The use of getLocaltime has been dropped with [openwrt/luci@68815c8](https://github.com/openwrt/luci/commit/68815c88b0ebcfd0116070510a4c43ccff1d0310) - seems you're running an incoherent package selection. Make sure that all LuCI packages, in particular luci-mod-system are updated to the latest version.

Awesome, thanks for the heads up!

closed 2022-10-15 by ihrapsa