ihrapsa/OctoWrt #21: Arc Welder - error: subprocess-exited-with-error

doczong, 2022-05-08:
I have tried various versions of the plugin from pip manually at the cmd line, and within octoprint.  All fail to install with no error other than the above.

It is after the setup.py legacy starts, or, after building wheels (if wheel is installed).  There is nothing in the logs or terminal to indicate which sub process.

Its probably related to compiling on the architecture as best guess.

Can provide a full copy and paste of the pip output if needed.

ihrapsa, 2022-05-08:
there's probably a building error of a dependency, yeah. Look at what's failing.

doczong, 2022-05-08:
Here's the full failure, to be honest, I am not sure what it is trying to say.  If it is referring to the most recent version of setuptools, I updated both it and PIP last night and both are at the latest available to it.

`root@OctoWrt:~# pip -v install https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/tags/1.1.0rc4.zip
Using pip 22.0.4 from /usr/lib/python3.9/site-packages/pip (python 3.9)
Collecting https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/tags/1.1.0rc4.zip
  Downloading https://github.com/FormerLurker/ArcWelderPlugin/archive/refs/tags/1.1.0rc4.zip
     - 289.5 kB 569.4 kB/s 0:00:00
  Running command python setup.py egg_info
  /tmp/pip-req-build-vn3zeb4_/setup.py:86: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
    if LooseVersion(octoprint.server.VERSION) < LooseVersion("1.4"):
  Found packages: {'octoprint_arc_welder_setuptools', 'octoprint_arc_welder', 'octoprint_arc_welder.test'}
  running egg_info
  creating /tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info
  writing /tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/PKG-INFO
  writing dependency_links to /tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/dependency_links.txt
  writing entry points to /tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/entry_points.txt
  writing requirements to /tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/requires.txt
  writing top-level names to /tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/top_level.txt
  writing manifest file '/tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/SOURCES.txt'
  reading manifest file '/tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no files found matching 'README.md'
  adding license file 'LICENSE'
  writing manifest file '/tmp/pip-pip-egg-info-izkr0eed/Arc_Welder.egg-info/SOURCES.txt'
  Preparing metadata (setup.py) ... done
Requirement already satisfied: six in /usr/lib/python3.9/site-packages (from Arc-Welder==1.1.0rc4) (1.16.0)
Requirement already satisfied: OctoPrint>1.3.8 in /usr/lib/python3.9/site-packages (from Arc-Welder==1.1.0rc4) (1.7.3)
Requirement already satisfied: setuptools>=6.0 in /usr/lib/python3.9/site-packages (from Arc-Welder==1.1.0rc4) (62.1.0)
Requirement already satisfied: colorlog<6,>=5.0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (5.0.1)
Requirement already satisfied: feedparser<7,>=6.0.8 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (6.0.8)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (2021.10.11)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (1.0.0)
Requirement already satisfied: regex in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (2022.1.18)
Requirement already satisfied: psutil<6,>=5.8 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (5.8.0)
Requirement already satisfied: pkginfo<2,>=1.7.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (1.8.2)
Requirement already satisfied: Jinja2<3,>=2.11.3 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (2.11.3)
Requirement already satisfied: pathvalidate<3,>=2.4.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (2.5.0)
Requirement already satisfied: sentry-sdk<2,>=1.3.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (1.5.11)
Requirement already satisfied: future<1,>=0.18.2 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (0.18.2)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (1.1.1)
Requirement already satisfied: blinker<2,>=1.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (1.4)
Requirement already satisfied: zeroconf<0.34,>=0.33 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (0.33.4)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (2021.2.23)
Requirement already satisfied: pyserial<4,>=3.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (3.5)
Requirement already satisfied: watchdog==0.10.4 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (0.10.4)
Requirement already satisfied: cachelib<0.2,>=0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (0.1.1)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (1.0.1)
Requirement already satisfied: wrapt<1.13,>=1.12.1 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (1.12.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (2.0)
Requirement already satisfied: netaddr<0.9,>=0.8 in /usr/lib/python3.9/site-packages (from OctoPrint>1.3.8->Arc-Welder==1.1.0rc4) (0.8.0)
[... cut ...]
  '"'"''"'"''"'"' % ('"'"'/tmp/pip-req-build-vn3zeb4_/setup.py'"'"',), "<pip-setuptools-caller>", "exec"))' install --record /tmp/pip-record-iujw356k/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.9/Arc-Welder
  cwd: /tmp/pip-req-build-vn3zeb4_/
  Running setup.py install for Arc-Welder ... error
error: legacy-install-failure

× Encountered error while trying to install package.
╰─> Arc-Welder

note: This is an issue with the package mentioned above, not pip.
hint: See above for output from the failure.`

ihrapsa, 2022-05-08:
oh my, try inserting the log between "```" next time for better formatting

It seems to be the Cython code that can't be compiled on the box (no proper compiler for mips arch). I think it needs to be cross compiled and that might be the only way.

doczong, 2022-05-08:
LOL, yup, I tried that double `s or triples strikeout the text.

Anyhow, that;s what I thought.  PITA.

closed 2022-10-15 by ihrapsa