ihrapsa/OctoWrt #23: Access Octowrt page

flake32, 2022-05-22:
Hi! after Octowrt and Octoprint successfull instalations I can access both if I connect directly to Octowrt wifi network, but if I connect via the client network I cant.
I see the lease on the router and I can ping it, but I try to acces the gui  it gives an ERR_CONNECTION_REFUSED error for both Octowrt and Octoprint.

Thanks in advance

ihrapsa, 2022-05-23:
> Hi! after Octowrt and Octoprint successfull instalations I can access both if I connect directly to Octowrt wifi network, but if I connect via the client network I cant. I see the lease on the [...]

you sure it's on `http` and not `https`?

pFallenAngelq, 2022-09-30:
Same problem here. Tryed almost every troubleshooting found on da webz including modifying the hosts file on drivers/etc but can't make it work. Just refuses the connection.

ihrapsa, 2022-09-30:
> Same problem here. Tryed almost every troubleshooting found on da webz including modifying the hosts file on drivers/etc but can't make it work. Just refuses the connection.

Hi, I don't know what you've done wrong but I'll try to reproduce your issue. Have you tried starting over with the guide? Once you setup wifi client connect to that new ip instead and continue with the guide

Swarmix23, 2023-02-05:
Hello,
I managed to successfully install extroot and octoprint script in Creality box, but I don't know how to do step number 4

"4. Access Octoprint UI on port 5000
http://box-ip:5000"

I can access the OpenWrt web interface via WWAN, can I find the IP address of the Creality box in Interfaces LAN? I try to enter it also with :5000, but a timeout occurs. I even tried to change the LAN IP address, but still the same.

Do you know what to do? Thank you very much!!!