ihrapsa/OctoWrt #25: WebCam Help for a complete Noob

BigAndra, 2022-06-08:
I have managed to make my way through the setup with a bit of help from youtube. Everything is all working well, so thanks. However, I cannot figure out how to get the webcam working - any chance you can explain (like you would to a small child) how to set this up?

ihrapsa, 2022-06-09:
What webcam is that? If you plug it in, can you access the webcam streaming server on `http://your-box-ip:8080`?

Neilridley, 2022-06-09:
ive same issue, not accessible on :8080 and doesn't show a stream when you input http://my-box ip:8080/?action=stream)

ihrapsa, 2022-06-09:
"Basic 2k camera" is not a useful information. As long as it has a `mjpg` stream it will work. Check with `v4l2-ctl --list-formats` in terminal. Send the output o that
Also, watch `logread -f` while plugging the camera, the `mjpg-streamer` service should automatically start. Send the output of that as well.

Neilridley, 2022-06-09:
sorry there was little info written on it,[ ](https://www.google.co.uk/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwj85_qDq6H4AhXPRMAKHak5DiwQFnoECAgQAQ&url=https%3A%2F%2Fwww.amazon.co.uk%2FMicrophone-Hiievpu-Suitable-Streaming-Conferencing%2Fdp%2FB08H1L41KF&usg=AOvVaw3rI8JnIAnBgrlRcPqTaKoO)

Neilridley, 2022-06-10:
i tried rerunning package and got the following
opkg update
opkg install gcc make unzip htop wget-ssl git-http
opkg install v4l-utils mjpg-streamer-input-uvc mjpg-streamer-output-http mjpg-streamer-www 

theCollected errors:
 * opkg_install_cmd: Cannot install package mjpg-streamer-input-uvc.
 * opkg_install_cmd: Cannot install package mjpg-streamer-output-http.
 * opkg_install_cmd: Cannot install package mjpg-streamer-www.

Neilridley, 2022-06-10:
this is what i get on logread -fFri Jun 10 06:56:44 2022 kern.info kernel: [  619.056712] usb 1-1.2: USB disconnect, device number 6
Fri Jun 10 06:56:53 2022 kern.info kernel: [  628.495687] usb 1-1.2: new high-speed USB device number 7 using ehci-platform
Fri Jun 10 06:56:54 2022 kern.warn kernel: [  628.624879] usb 1-1.2: config 1 interface 0 altsetting 0 endpoint 0x83 has an invalid bInterval 32, changing to 9
Fri Jun 10 06:56:54 2022 kern.info kernel: [  628.663980] uvcvideo: Found UVC 1.00 device GENERAL WEBCAM (1b3f:1167)
Fri Jun 10 06:56:54 2022 kern.err kernel: [  628.684807] uvcvideo: Failed to query (GET_INFO) UVC control 2 on unit 1: -32 (exp. 1).
Fri Jun 10 06:56:54 2022 kern.info kernel: [  628.728428] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
Fri Jun 10 06:56:54 2022 kern.info kernel: [  628.761539] input: GENERAL WEBCAM: GENERAL WEBCAM as /devices/platform/101c0000.ehci/usb1/1-1/1-1.2/1-1.2:1.0/input/input3

ihrapsa, 2022-06-10:
> i tried rerunning package and got the following opkg update opkg install gcc make unzip htop wget-ssl git-http opkg install v4l-utils mjpg-streamer-input-uvc mjpg-streamer-output-http [...]

Do not strip teh errror log. Send it whole. Does the box have internet access?

Neilridley, 2022-06-10:
Yes.box has internet, I’m away for weekend so will send log Sunday, thanks for your help

theolensky, 2022-06-12:
> i tried rerunning package and got the following opkg update opkg install gcc make unzip htop wget-ssl git-http opkg install v4l-utils mjpg-streamer-input-uvc mjpg-streamer-output-http [...]

I am facing the exact same issue with a Aoboco 1080p camera. I've been trying to get the webcam stream to work for days but it doesnt seem to. Let me know if you have had any progress?

Neilridley, 2022-06-13:
> Do not strip teh errror log. Send it whole. Does the box have internet access?

full log

root@OctoWrt:~# opkg update
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ramips/mt76x8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ramips/mt76x8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
root@OctoWrt:~# opkg install gcc make unzip htop wget-ssl git-http
Package gcc (7.4.0-5) installed in root is up to date.
Package make (4.3-1) installed in root is up to date.
Package unzip (6.0-8) installed in root is up to date.
Package htop (3.2.1-1) installed in root is up to date.
Package wget-ssl (1.21.1-1) installed in root is up to date.
Package git-http (2.34.3-1) installed in root is up to date.
root@OctoWrt:~# opkg install v4l-utils mjpg-streamer-input-uvc mjpg-streamer-output-http mjpg-streamer-www
Package v4l-utils (1.20.0-3) installed in root is up to date.
Unknown package 'mjpg-streamer-input-uvc'.
Unknown package 'mjpg-streamer-output-http'.
Unknown package 'mjpg-streamer-www'.
Collected errors:
 * opkg_install_cmd: Cannot install package mjpg-streamer-input-uvc.
 * opkg_install_cmd: Cannot install package mjpg-streamer-output-http.
 * opkg_install_cmd: Cannot install package mjpg-streamer-www.
root@OctoWrt:~#

Neilridley, 2022-06-13:
root@OctoWrt:~# logread -f
Mon Jun 13 14:10:42 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:42,402 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2022.3.28) did not pass check, not loading.
Mon Jun 13 14:10:48 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:48,627 - octoprint.plugin.core - INFO - Found 19 plugin(s) providing 19 mixin implementations, 43 hook handlers
Mon Jun 13 14:10:51 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:51,292 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
Mon Jun 13 14:10:51 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:51,586 - octoprint.server - INFO - Intermediary server started
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,226 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /.octoprint/uploads...
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,253 - octoprint.filemanager.storage - INFO - ... file metadata for /.octoprint/uploads initialized successfully.
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,353 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,359 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,398 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,423 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,449 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,476 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,487 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_GRANT (needs: "Need(method='role', value='plugin_appkeys_user')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,524 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,549 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,575 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,604 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_LIST (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,627 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,652 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_manage'), Need(method='role', value='plugin_pluginmanager_install')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,676 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,699 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
Mon Jun 13 14:10:52 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:52,706 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
Mon Jun 13 14:10:55 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:55,081 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
Mon Jun 13 14:10:55 2022 daemon.info octoprint[1813]: 2022-06-13 14:10:55,265 - octoprint.plugins.softwareupdate - INFO - Loaded update log from disk
Mon Jun 13 14:11:28 2022 daemon.info octoprint[1813]: 2022-06-13 14:11:28,516 - octoprint.util.pip - INFO - Using "/usr/bin/python3 -m pip" as command to invoke pip
Mon Jun 13 14:11:47 2022 kern.info kernel: [  187.187890] usb 1-1.4: USB disconnect, device number 4

ihrapsa, 2022-06-13:
> 

Did you do the `opkg update` before installing the mjpg-streamer packages?

Neilridley, 2022-06-13:
yes thats the order they were in on the install page

ihrapsa, 2022-06-14:
> yes thats the order they were in on the install page
Did you use the manual or automatic steps?

Neilridley, 2022-06-14:
> yes thats the order they were in on the install page
> Did you use the manual or automatic steps?

automatic on initial install then manual when web cam didn't work

s1lvi0, 2022-06-14:
I have the same problem, I have used the automatic method but the webcam didn't work. The I tried to manual install mjpg-streamer but this is the response:

```
BusyBox v1.33.2 (2022-02-13 17:28:05 UTC) built-in shell (ash)

  _______        __          ________        __   
 |       |.----.|  |_.-----.|  |  |  |.----.|  |_ 
 |   -   ||  __||   _|  _  ||  |  |  ||   _||   _|
 |_______||____||____|_____||________||__|  |____|
     O C T O P R I N T   O N   O P E N W R T 
 -----------------------------------------------------
 OpenWrt 21.02-SNAPSHOT, r16490-781b5d8f30
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OctoWrt:~# opkg update
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ramips/mt76x8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/targets/ramips/mt76x8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/21.02-SNAPSHOT/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
root@OctoWrt:~# opkg install mjpg-streamer-input-uvc
Unknown package 'mjpg-streamer-input-uvc'.
Collected errors:
 * opkg_install_cmd: Cannot install package mjpg-streamer-input-uvc.
root@OctoWrt:~# 

```

Neilridley, 2022-06-17:
Has anyone got the webcam to work yet or is it dead in the water?

ihrapsa, 2022-06-17:
It seems like the repo package lists broke again (probably cause the box is using snapshot OpenWrt images since it's not officially supported yet). I haven't looked into a fix yet but you could try testing in the meantime with the `distfeed fix` from [here](https://github.com/ihrapsa/OctoWrt#%EF%B8%8F-preparing)

Neilridley, 2022-06-17:
I gave it a try but no change

Renmike80, 2022-06-24:
I am facing the same issue after installing OctoWrt/OctoPrint earlier today. Is there anyplace else that the images can be acquired from? I searched (starting with OpenWrt) and haven't found anything consistent/reliable to try yet. One of the reasons I wanted to give OctoPrint/OctoWrt a try was to get my webcam to stream my print activity.

ihrapsa, 2022-06-24:
> I am facing the same issue after installing OctoWrt/OctoPrint earlier today. Is there anyplace else that the images can be acquired from? I searched (starting with OpenWrt) and haven't found [...]

As stated above the issue hasn't been resolved yet. I'll try to build a new updated OpenWrt image this weekend. 

Note that you don't need octoprint just to have the print streamed

duchengyao, 2022-06-25:
I am facing the same issue.  And I have another question. Does the creality box have enough performance to support the timelapse plug-in?

ihrapsa, 2022-06-25:
> I am facing the same issue. And I have another question. Does the creality box have enough performance to support the timelapse plug-in?

I've been succesful creating timelapses with the box but encoding the frames into a movie is not fast at all. If you really want timelapses, disable rendering and do it on a more powerfull computer.

duchengyao, 2022-06-25:
Thanks for reply. Is there any way to install those `mjpg-streamer` package without using new `OpenWrt image`? It takes too long to reinstall octoprint.

I have tried dist for `21.02.0` and `21.02.3`, but still cannot find those package.

ihrapsa, 2022-06-25:
> Thanks for reply. Is there any way to install those `mjpg-streamer` package without using new `OpenWrt image`? It takes too long to reinstall octoprint.
> 
> I have tried dist for `21.02.0` and `21.02.3`, but still cannot find those package.
> 
> ================
> 
> I finally installed those package from `https://downloads.openwrt.org/releases/22.03.0-rc4/`
> 
> But, it seems like linked to wrong library.
> 
> ```
> root@KlipperWrt:/tmp/log# /usr/bin/mjpg_streamer
> Error relocating /usr/bin/mjpg_streamer: __dlsym_time64: symbol not found
> ```

try doing the distfeed fix from [here.](https://github.com/ihrapsa/OctoWrt#%EF%B8%8F-preparing) I've changed the lists to snapshot. It works on my tests

Renmike80, 2022-06-25:
I am trying the updated distfeed and get the following error when I try to add the repositories needed for mjpg-streamer:

`root@OctoWrt:~# opkg install v4l-utils mjpg-streamer-input-uvc mjpg-streamer-output-http mjpg-streamer-www
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (librt and librt) providing same name marked HOLD or PREFER. Using latest.
Multiple packages (libstdcpp6 and libstdcpp6) providing same name marked HOLD or PREFER. Using latest.
Upgrading v4l-utils on root from 1.20.0-3 to 1.20.0-4...
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/v4l-utils_1.20.0-4_mipsel_24kc.ipk
Error relocating /usr/bin/wget: __gmtime64: symbol not found
Error relocating /usr/bin/wget: __localtime64_r: symbol not found
Error relocating /usr/bin/wget: __select_time64: symbol not found
Error relocating /usr/bin/wget: __setitimer_time64: symbol not found
Error relocating /usr/bin/wget: __lstat_time64: symbol not found
Error relocating /usr/bin/wget: __clock_getres_time64: symbol not found
Error relocating /usr/bin/wget: __localtime64: symbol not found
Error relocating /usr/bin/wget: __stat_time64: symbol not found
Error relocating /usr/bin/wget: __fstat_time64: symbol not found
Error relocating /usr/bin/wget: __utime64: symbol not found
Error relocating /usr/bin/wget: __clock_gettime64: symbol not found
Error relocating /usr/bin/wget: __gmtime64_r: symbol not found
Error relocating /usr/bin/wget: __time64: symbol not found
Multiple packages (libpthread and libpthread) providing same name marked HOLD or PREFER. Using latest.
Installing mjpg-streamer-input-uvc (1.0.0-5) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/mjpg-streamer-input-uvc_1.0.0-5_mipsel_24kc.ipk
Error relocating /usr/bin/wget: __gmtime64: symbol not found
Error relocating /usr/bin/wget: __localtime64_r: symbol not found
Error relocating /usr/bin/wget: __select_time64: symbol not found
Error relocating /usr/bin/wget: __setitimer_time64: symbol not found
Error relocating /usr/bin/wget: __lstat_time64: symbol not found
Error relocating /usr/bin/wget: __clock_getres_time64: symbol not found
Error relocating /usr/bin/wget: __localtime64: symbol not found
Error relocating /usr/bin/wget: __stat_time64: symbol not found
Error relocating /usr/bin/wget: __fstat_time64: symbol not found
Error relocating /usr/bin/wget: __utime64: symbol not found
Error relocating /usr/bin/wget: __clock_gettime64: symbol not found
Error relocating /usr/bin/wget: __gmtime64_r: symbol not found
Error relocating /usr/bin/wget: __time64: symbol not found
Installing mjpg-streamer-output-http (1.0.0-5) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/mjpg-streamer-output-http_1.0.0-5_mipsel_24kc.ipk
Error relocating /usr/bin/wget: __gmtime64: symbol not found
Error relocating /usr/bin/wget: __localtime64_r: symbol not found
Error relocating /usr/bin/wget: __select_time64: symbol not found
Error relocating /usr/bin/wget: __setitimer_time64: symbol not found
Error relocating /usr/bin/wget: __lstat_time64: symbol not found
Error relocating /usr/bin/wget: __clock_getres_time64: symbol not found
Error relocating /usr/bin/wget: __localtime64: symbol not found
Error relocating /usr/bin/wget: __stat_time64: symbol not found
Error relocating /usr/bin/wget: __fstat_time64: symbol not found
Error relocating /usr/bin/wget: __utime64: symbol not found
Error relocating /usr/bin/wget: __clock_gettime64: symbol not found
Error relocating /usr/bin/wget: __gmtime64_r: symbol not found
Error relocating /usr/bin/wget: __time64: symbol not found
Installing mjpg-streamer-www (1.0.0-5) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/mjpg-streamer-www_1.0.0-5_mipsel_24kc.ipk
Error relocating /usr/bin/wget: __gmtime64: symbol not found
Error relocating /usr/bin/wget: __localtime64_r: symbol not found
Error relocating /usr/bin/wget: __select_time64: symbol not found
Error relocating /usr/bin/wget: __setitimer_time64: symbol not found
Error relocating /usr/bin/wget: __lstat_time64: symbol not found
Error relocating /usr/bin/wget: __clock_getres_time64: symbol not found
Error relocating /usr/bin/wget: __localtime64: symbol not found
Error relocating /usr/bin/wget: __stat_time64: symbol not found
Error relocating /usr/bin/wget: __fstat_time64: symbol not found
Error relocating /usr/bin/wget: __utime64: symbol not found
Error relocating /usr/bin/wget: __clock_gettime64: symbol not found
Error relocating /usr/bin/wget: __gmtime64_r: symbol not found
Error relocating /usr/bin/wget: __time64: symbol not found
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/v4l-utils_1.20.0-4_mipsel_24kc.ipk, wget returned 127.
 * opkg_install_pkg: Failed to download v4l-utils. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package v4l-utils.
[... cut ...]
 * opkg_install_pkg: Failed to download mjpg-streamer-output-http. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package mjpg-streamer-output-http.
 * opkg_download: Failed to download https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/mjpg-streamer-www_1.0.0-5_mipsel_24kc.ipk, wget returned 127.
 * opkg_install_pkg: Failed to download mjpg-streamer-www. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package mjpg-streamer-www.`

duchengyao, 2022-06-26:
> try doing the distfeed fix from [here.](https://github.com/ihrapsa/OctoWrt#%EF%B8%8F-preparing) I've changed the lists to snapshot. It works on my tests

Still linked to wrong library `Error relocating /usr/bin/mjpg_streamer: __dlsym_time64: symbol not found`

closed 2022-10-15 by ihrapsa