ihrapsa/OctoWrt #26: how can i navigate the internal files of the system?

TheFr0st3r, 2022-06-18:
excellent version, but in the previous one there was in the "services" box the possibility of scrolling through the files of the system, in addition to the already present "terminal", is it possible to re-insert it?

ihrapsa, 2022-06-19:
> excellent version, but in the previous one there was in the "services" box the possibility of scrolling through the files of the system, in addition to the already present "terminal", is it possible to re-insert it?

install `luci-app-ttyd` for terminal and [this](https://github.com/ihrapsa/luci-app-file_transfer/) for file browser.

TheFr0st3r, 2022-06-19:
Thank you! I noticed however that the webcam does not work because of the lack of the file that precisely needs, how can I install it? (the mjpg-streamer file is missing due to an error during the installation that I did not understand well)

ihrapsa, 2022-06-19:
That's an issue I'm aware of and it's probably because the bin is still a snapshot and not an official binary. I'll try to fix it when i get some free time.

TheFr0st3r, 2022-06-19:
uh ok! Thank you

TheFr0st3r, 2022-06-19:
a question for the installation of the file manager, do I have to enter on teminale this command as well as? "luci-app-ttyd" or do I have something different to do?

ihrapsa, 2022-06-19:
> a question for the installation of the file manager, do I have to enter on teminale this command as well as? "luci-app-ttyd" or do I have something different to do?

use the luci web ui to install the package or the ssh terminal to install it as any other OpenWrt package

TheFr0st3r, 2022-06-19:
the package is already installed and already updated, so I believe that I have to do something for the browser, what should I do? I'm not really easy going about it

ihrapsa, 2022-06-19:
what package is already installed? if the file browser package luci app is not in the repo you will probably need to download the `.ipk` file and manually intstall it with `opkg install <package_name>.ipk`

TheFr0st3r, 2022-06-19:
writing the command "opkg install luci-app-ttyd" reports that it is already installed

ihrapsa, 2022-06-19:
> writing the command "opkg install luci-app-ttyd" reports that it is already installed

Then it's installed and it should be under Services tab in luci UI.

TheFr0st3r, 2022-06-19:
no there isn't, could it be another missing file as for the "mjpg-streamer"?

ihrapsa, 2022-06-19:
I don't know what you mean by "another missing file as for the "mjpg-streamer". Give me more info, install logs, try rebooting the box, try removing then reinstalling the package.

ihrapsa, 2022-06-19:
If this is really a hassle for you try using a sftp program like winscp for filebrowsing on the box. For terminal you can use Powershell in windows or any terminal in UNIX systems then use `ssh root@192.168.x.x` to ssh into the box's terminal`

TheFr0st3r, 2022-06-20:
yes thank you, in the end I used WinSCP, to the rest everything affixed! I hope you will solve the problem with the webcam soon!

closed 2022-10-15 by ihrapsa