ihrapsa/OctoWrt #27: page not found

Darkstar83, 2022-07-10:
❗ Check Releases page for binaries ❗

there is no such page, 

I am completeley new to Github, i cannot find the Releases page, Either the page is not there or i do not know how to navigate to said page

any assistance will be appreciated.

ihrapsa, 2022-07-10:
it's right there on the right side panel of the main page. https://github.com/ihrapsa/OctoWrt/releases/tag/v1.0.0

Darkstar83, 2022-07-10:
May I ask further questions if I get stuck somewhere through the install?

Darkstar83, 2022-07-10:
How do I SSH onto it:  achieve step 4?
(https://github.com/ihrapsa/OctoWrt/tree/main/firmware)/OpenWrt_snapshot/

ihrapsa, 2022-07-10:
> How do I SSH onto it: achieve step 4? (https://github.com/ihrapsa/OctoWrt/tree/main/firmware)/OpenWrt_snapshot/

for internet setup follt this: https://github.com/ihrapsa/OctoWrt#1-flash-openwrt-following-guide-here
no need to ssh. everything can be done using luci UI

ihrapsa, 2022-07-10:
Io don't need to ssh anymore. Internet setup can be dune through luvi web ui. follow this: 
https://github.com/ihrapsa/OctoWrt#1-flash-openwrt-following-guide-here

ihrapsa, 2022-07-12:
It's kind of a mess what you're doing there. What klipper firmware are you talking about? you can't flash klipper on the box. The box runs OpenWrt. Since the issue is opened on the OctoWrt page I reckon that you want to install octoprint. Did you follow the steps from the guide?

Darkstar83, 2022-07-15:
Klipper with fluidd

ihrapsa, 2022-07-16:
> Klipper with fluidd

This guide is for Octoprint. Check KlipperWrt repo for that

Darkstar83, 2022-07-16:
Ok Thank you very much for all the help given. Further help will be obtained from the Klipper Repositories. Have a blessed day.

ihrapsa, 2022-07-16:
you too! check out the discord server for further help

closed 2022-07-16 by ihrapsa