ihrapsa/OctoWrt #29: Problem config camera

Pepitoo597, 2022-08-17:
Hello I would like to configure my Logitech C160 camera which is compatible with mjpg-streamer but the camera lags so I would like to solve the problem but I don't know which file to go to please

ihrapsa, 2022-08-18:
Edit fps line inside `/etc/config/mjpg-streamer` file. Don't go higher than 15fps  if you use the box to stream gcode as well. Or experiment with it

Pepitoo597, 2022-08-18:
Ok thank you, after testing my camera still has this latency problem, I think it is better to change the camera. Do you have a camera to recommend to me?

ihrapsa, 2022-08-18:
Wait, high latency or low framerate? Two different things.

Pepitoo597, 2022-08-18:
I don't really know how to explain it in fact when the printer makes a movement the camera will take a while before seeing the printer move it is not fluid.

ihrapsa, 2022-08-18:
check the difference between latency and framerate. what framerate did you set inside mjpg-streamer file?

Pepitoo597, 2022-08-18:
I set the framerate to 15

ihrapsa, 2022-08-18:
do `cat /etc/config/mjpg-streamer`and show me

Pepitoo597, 2022-08-18:
[image]

ihrapsa, 2022-08-18:
Frame rate is ok. Can you do short video with printer and stream at the same time to see what's the issue?

closed 2022-10-15 by ihrapsa