ihrapsa/OctoWrt #30: Octoprint crash loop

azlim16, 2022-10-13:
Hi
After setting wifi and running the first and second script without issue
rebooted a box and nothing happens
logged in open wrt and in the log the is an arror:
daemon.info procd: Instance octoprint::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

i tried reformatting the microsd and redoing the scripts many time but nothing,
I tried to restore a creality box with stock firmware and after that i reflashed openwrt firmware also without any luck

Suggestion?

here is the complete log:

https://pastebin.com/dbHwuVTE

ihrapsa, 2022-10-13:
Hi, these kernel log contents are not really useful. An install log would be more useful. I wonder if a new openwrt snapshot needs to be compiled as these have limited support. I'll have to reproduce the issue and see what's going on. Can you update opkg without any errors?

ihrapsa, 2022-10-15:
Fixed the install issue in [decef31](https://github.com/ihrapsa/OctoWrt/commit/decef312a690323730f90f10a27d4118f0df8626). The script has been updated to OctoPrint 1.8.4 as well.

closed 2022-10-15 by ihrapsa