ihrapsa/OctoWrt #31: Upgrade from 1.8.2 to Octoprint 1.8.4

laushere, 2022-10-16:
Hello community,

after realizing that the install scripts have been updated in order to install version 1.8.4, I´ve already tried an upgrade install. The upgrade from Octoprint 1.8.2 to 1.8.4 from the webinterface never worked. That´s why I already started to create screenshots yesterday in order to post an issue... Seems that someone heart my voice :-)

I´ve executed both scrips from the Luci terminal. I didnt throw any error. BUT after rebooting, Octoprint 1.8.2 re-appears?! 
Is there any working upgrade procedure or any option for a clean install?

ihrapsa, 2022-10-16:
You sure there were no errors? You don't need to execute both scriptat. First script only expands storage which you already did. You can alternatively uninstall curent Octoprint with pip then use the second script to install 1.8.4

laushere, 2022-10-16:
Uninstall worked fine:

root@octoprint:~# pip uninstall Octoprint
Found existing installation: OctoPrint 1.8.2
Uninstalling OctoPrint-1.8.2:
  Would remove:
    /usr/bin/octoprint
    /usr/lib/python3.10/site-packages/OctoPrint-1.8.2-py3.10.egg-info
    /usr/lib/python3.10/site-packages/octoprint/*
    /usr/lib/python3.10/site-packages/octoprint_client/*
    /usr/lib/python3.10/site-packages/octoprint_setuptools/*
  Would not remove (might be manually added):
    /usr/lib/python3.10/site-packages/octoprint/templates/_data/AUTHORS.md
    /usr/lib/python3.10/site-packages/octoprint/templates/_data/SUPPORTERS.md
    /usr/lib/python3.10/site-packages/octoprint/templates/_data/THIRDPARTYLICENSES.md
Proceed (Y/n)? y
  Successfully uninstalled OctoPrint-1.8.2

But during install, the following error appeared:
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/base.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/browser.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/connection.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/control.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/files.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/index.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/job.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/languages.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/logs.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/printer.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/printerprofiles.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/settings.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/slicing.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/socket.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/system.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/timelapse.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/users.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/util.rst  
  inflating: /root/OctoPrint-1.8.4/docs/jsclientlib/wizard.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/cli.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/filemanager.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/index.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/plugin.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/printer.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/server.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/settings.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/slicing.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/users.rst  
  inflating: /root/OctoPrint-1.8.4/docs/modules/util.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/concepts.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/controlproperties.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/distributing.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/gettingstarted.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/helpers.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/hooks.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/index.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/injectedproperties.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/mixins.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/python3_migration.rst  
  inflating: /root/OctoPrint-1.8.4/docs/plugins/viewmodels.rst  
  inflating: /root/OctoPrint-1.8.4/docs/sphinxext/codeblockext.py  
  inflating: /root/OctoPrint-1.8.4/docs/sphinxext/onlineinclude.py  
  inflating: /root/OctoPrint-1.8.4/pytest.ini  
  inflating: /root/OctoPrint-1.8.4/requirements.txt  
  inflating: /root/OctoPrint-1.8.4/scripts/README.md  
  inflating: /root/OctoPrint-1.8.4/scripts/octoprint.default  
  inflating: /root/OctoPrint-1.8.4/scripts/octoprint.init  
  inflating: /root/OctoPrint-1.8.4/scripts/octoprint.service  
  inflating: /root/OctoPrint-1.8.4/setup.cfg  
  inflating: /root/OctoPrint-1.8.4/setup.py  
  inflating: /root/OctoPrint-1.8.4/src/octoprint/__init__.py  
[... 948 lines cut ...]
Length: 30280 (30K) [application/octet-stream]
Saving to: '/tmp/ffmpeg/shine_3.1.1-1_mipsel_24kc.ipk'

shine_3.1.1-1_mipsel_24kc.ipk      100%[=============================================================>]  29.57K  --.-KB/s    in 0.1s    

2022-10-16 18:14:29 (219 KB/s) - '/tmp/ffmpeg/shine_3.1.1-1_mipsel_24kc.ipk' saved [30280/30280]

Package alsa-lib (1.2.4-1) installed in root is up to date.
Package fdk-aac (2.0.1-4) installed in root is up to date.
Package ffmpeg (4.3.2-1) installed in root is up to date.
Package ffprobe (4.3.2-1) installed in root is up to date.
Package libatomic1 (8.4.0-3) installed in root is up to date.
Package libbz21.0 (1.0.8-1) installed in root is up to date.
Package libffmpeg-full (4.3.2-1) installed in root is up to date.
Package libgmp10 (6.2.1-1) installed in root is up to date.
Package libgnutls (3.7.2-1) installed in root is up to date.
Package libnettle8 (3.6-1) installed in root is up to date.
Package libx264 (2020-10-26-1) installed in root is up to date.
Package shine (3.1.1-1) installed in root is up to date.
 
##################################
### Reboot and wait a while... ###
##################################
 
Press [ENTER] to reboot...or [ctrl+c]

ihrapsa, 2022-10-16:
I think I've found the issue. I've updated the script.

laushere, 2022-10-16:
Still the same issue:

/usr/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3.10/site-packages/setuptools/command/easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
error: error in 'egg_base' option: 'src' does not exist or is not a directory

ihrapsa, 2022-10-16:
do this in a ssh terminal 

```
cd /root/OctoPrint-1.8.4
python /root/OctoPrint-1.8.4/setup.py install -v
```

laushere, 2022-10-16:
Yeah, Setup finished successfully.
But Octoprint doesn’t accept my credentials any longer. 
Is there a command to delete all previous settings?

ihrapsa, 2022-10-16:
you can delete all octorpint data inside `/.octoprint` you would probably need to restart octoprint after that: `/etc/init.d/octoprint restart`

laushere, 2022-10-17:
Oh yeah, it´s starting again with the intial setup :-)

closed 2022-10-22 by ihrapsa