ihrapsa/OctoWrt #32: i can't download anything

Quarry2000, 2022-11-07:
i enter the very first commands,
cd /tmp
wget https://github.com/ihrapsa/octoprint/raw/main/scripts/1_format_exroot.sh
end then i get this respond:
Downloading 'https://github.com/ihrapsa/OctoWrt/raw/main/scripts/1_format_extroot.sh'
Failed to send request: Operation not permitted
root@OctoWrt:/tmp# 
i dont know what to do next

ihrapsa, 2022-11-07:
do you have internet connection on the box?

Quarry2000, 2022-11-07:
I think I have one, I connected my internet connection to it
Should I try using my phone connection?

Tamaș Alexandru Bela

ihrapsa, 2022-11-07:
you need to be sure you have internet connection first. Do some `ping` tests and report

ihrapsa, 2022-11-07:
What did you do up until now? Are you connected through the ip your router assigned?

Quarry2000, 2022-11-07:
I'm connected through the ip I assigned
192.168.1.1 was already taken, so I assigned it 192.168.1.99 and I changed name and pass
I added a Ethernet to communicate with pc at first and then connected through WiFi

Tamaș Alexandru Bela

Quarry2000, 2022-11-07:
I was using a guide on YouTube to work it out

Tamaș Alexandru Bela

ihrapsa, 2022-11-07:
I'm not sure I understand what you did. You don't have to change the static ip unless your router is on the same subnet (i.e. when your router is also on 192.168.1.1), in which case you change the subnet (from 192.168.1.1 to 192.168.x.1 where x is different than 1). No need to change the host part of the ip (the last part). After that you need to set up the box as a client to your router. You router will work out the ip based on its dhcp server. I don't think you set up network correctly. Follow the guide carefully and continue with the two scripts only after you are able to access the box on your home network (by using the ip your router asigns)

Quarry2000, 2022-11-07:
Thank you very much
I will follow

Tamaș Alexandru Bela