ihrapsa/OctoWrt #36: OctoPrint not starting

tofuSCHNITZEL, 2022-12-06:
Hi,
after following the steps the octoprint Webgui was not reachable under port 5000 so I went into ssh and saw that octoprint service is stopped, so tried to run it manually and got these errors:
`2022-12-06 13:36:29,383 - octoprint.startup - CRITICAL - Could not initialize settings manager: (urllib3 2.0.0a2 (/usr/lib/python3.10/site-packages/urllib3-2.0.0a2-py3.10.egg), Requirement.parse('urllib3<1.27,>=1.21.1'), {'requests'})
2022-12-06 13:36:29,398 - octoprint.startup - CRITICAL - There was a fatal error starting up OctoPrint.
Could not initialize settings manager: (urllib3 2.0.0a2 (/usr/lib/python3.10/site-packages/urllib3-2.0.0a2-py3.10.egg), Requirement.parse('urllib3<1.27,>=1.21.1'), {'requests'})
There was a fatal error starting up OctoPrint.`

then I tried running `python /root/OctoPrint-1.8.4/setup.py install ` again and got:

`error: urllib3 2.0.0a2 is installed but urllib3<1.27,>=1.21.1 is required by {'requests'}`

I tried `pip install urllib3==1.21.1` but it did not help and I could not find urllib dependency in setup.py or requirements.txt

any hints?

ihrapsa, 2022-12-06:
Hi, try installing the openwrt precompiled package: `opkg update && opkg install python3-requests`

tofuSCHNITZEL, 2022-12-06:
thanks for the tip... it still gave me the same error so I ran `pip uninstall urllib3` multiple times and then `opkg install --force-reinstall python3-requests python3-urllib3` but then it just said "The 'urllib3>=1.26.11' distribution was not found" trying to install urllib3 with pip with version 1.26.11 also resulted in more conflicted dependencies...

BCorbeek, 2023-01-15:
@tofuSCHNITZEL Hi! How did you end up fixing this exactly? I'm having the exact same problem - it's unclear what your final fix is. 

pip uninstall urllib -> install all packages -> run setup.py?

tofuSCHNITZEL, 2023-01-15:
Sorry, not sure anymore.
But yes it was something along removing the pip version of the package (requests and urllib) and installing the package via opkg then run setup.py again so you see if it works.

BCorbeek, 2023-01-15:
Ah missed that link - will try that, thanks!