ihrapsa/OctoWrt #37: Manual install fails

krystof018, 2022-12-19:
When I run `opkg update && opkg install swap-utils zram-swap` I get this as output:

```
Downloading https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Package swap-utils (2.38-2) installed in root is up to date.
Installing zram-swap (32) to root...
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/zram-swap_32_all.ipk
Configuring zram-swap.
zram_start: activating '/dev/zram0' for swapping (59 MiB)
zram_reset: enforcing defaults via /sys/block/zram0/reset
zram_comp_algo: set compression algorithm 'lzo' for zram '/dev/zram0'
Setting up swapspace version 1, size = 61861888 bytes
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.10.159-1-7d75de85a1f4cde1b2401b5c815e0218) for kmod-zram

```

ihrapsa, 2022-12-20:
You can skip the zram-swap since there’s no noticable performance improvement. Is there any reason to why you are following the manual steps?

krystof018, 2022-12-20:
I was trying manual steps because automatic install fails to install some packages and octoprint is then not working.

`Collected errors:
 * opkg_install_cmd: Cannot install package python3-psutil.
 * pkg_hash_check_unresolved: cannot find dependency python3-pycparser for python3-cffi
 * pkg_hash_fetch_best_installation_candidate: Packages for python3-cffi found, but incompatible with the architectures configured
`

ihrapsa, 2022-12-20:
Are you on latest openwrt firmware?

b4ng4r4ng, 2022-12-20:
> I was trying manual steps because automatic install fails to install some packages and octoprint is then not working.
>
> ```
> Collected errors:
>  * opkg_install_cmd: Cannot install package [...]

I'm also encountering a similar issue with the automatic installation.
Before this, I encountered an issue with urllib3 like described in #36.
The code below resolved the urllib3 problem:
```sh
pip uninstall urllib3
pip install -Iv urllib3==1.26.11
opkg update && opkg install python3-requests
cd /root/OctoPrint-1.8.4
python setup.py install
```

However, setup.py fails with:
```sh
Processing argon2-cffi-bindings-21.2.0.tar.gz
Writing /tmp/easy_install-rbsbf6fd/argon2-cffi-bindings-21.2.0/setup.cfg
Running argon2-cffi-bindings-21.2.0/setup.py -q bdist_egg --dist-dir /tmp/easy_install-rbsbf6fd/argon2-cffi-bindings-21.2.0/egg-dist-tmp-05hay9tu
/usr/lib/python3.10/site-packages/setuptools/installer.py:27: SetuptoolsDeprecationWarning: setuptools.installer is deprecated. Requirements should be satisfied by a PEP 517 installer.
  warnings.warn(
WARNING: The wheel package is not available.
WARNING: The wheel package is not available.
  Preparing metadata (setup.py) ... done
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  Building wheel for cffi (setup.py) ... error
  ERROR: Failed building wheel for cffi
ERROR: Failed to build one or more wheels
error: Command '['/usr/bin/python', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/easy_install-rbsbf6fd/argon2-cffi-bindings-21.2.0/temp/tmpl_bzr6cy', '--quiet', 'cffi>=1.0.1']' returned non-zero exit status 1.
```

In an attempt to resolve the cffi issue, I tried the following:
```sh
opkg update && opkg install python3-cffi
```
But this returns the following error:
```sh
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency python3-pycparser for python3-cffi
 * pkg_hash_fetch_best_installation_candidate: Packages for python3-cffi found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package python3-cffi.
```

I noticed that pycparser isn't a part of all the OpenWrt packages.
@ihrapsa, I'm guessing that's where your question regarding firmware version might come from.

My version: `OpenWrt SNAPSHOT, r20027-586c2426d9` (from Release page)

b4ng4r4ng, 2022-12-28:
When I last worked on this, python3-pycparser was missing from:
https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/

Today I was able to run the command below and install python3-cffi without any problems, since it seems the dependent python3-pycparser package mysteriously reappeared. I guess that's the danger of these snapshot builds...
```shell
opkg update && opkg install python3-cffi
```
I then ran the following command and Octoprint is running sucessfully after a poweroff.
```shell
cd /root/OctoPrint-1.8.4
python setup.py install
```