ihrapsa/OctoWrt #39: OctoPrint won't start

LorenzoBonelli0102, 2022-12-27:
Hi, I'm reporting this issue after two days of trying to make this work.
I have tried to format and reinstall OpenWrt multiple times but after running the second script and rebooting it gets stuck with the same error every time.
This is the logread so maybe someone can figure out what's going on.

Tue Dec 27 12:32:01 2022 daemon.err octoprint[3049]: Could not initialize settings manager: (urllib3 2.0.0a2 (/usr/lib/python3.10/site-packages/urllib3-2.0.0a2-py3.10.egg), Requirement.parse('urllib3<1.27,>=1.21.1'), {'requests'})

Tue Dec 27 12:32:01 2022 daemon.err octoprint[3049]: There was a fatal error starting up OctoPrint.

Tue Dec 27 12:32:03 2022 daemon.info procd: Instance octoprint::instance1 s in a crash loop 6 crashes, 40 seconds since last crash

ihrapsa, 2022-12-27:
Hi, this has been reported before. Check #36

markhicken, 2022-12-28:
It has been reported before, but I am still running into this issue after a clean install and the previous suggested solution didn't help.

b4ng4r4ng, 2022-12-28:
@markhicken, @LorenzoBonelli0102
I also encountered the urllib3 issue (#36) and this is exactly what worked for me:
```shell
pip uninstall urllib3
pip install -Iv urllib3==1.26.11
opkg update && opkg install python3-requests
cd /root/OctoPrint-1.8.4
python setup.py install
```
Sometime last week there was an issue with the OpenWrt snapshot repository for this target architecture. There was a missing package which caused python3-cffi not to install correctly. (https://github.com/ihrapsa/OctoWrt/issues/37#issuecomment-1366837925)
So, if you've been fighting with these issues since last week (_like me_), you might want to try the following instead:
```shell
pip uninstall urllib3
pip install -Iv urllib3==1.26.11
opkg update
opkg install python3-requests
opkg install python3-cffi
cd /root/OctoPrint-1.8.4
python setup.py install
```

markhicken, 2022-12-29:
I started from scratch and for whatever reason, it didn't automatically add my micro sd card to the `/overlay` which meant that I would run out of space during the install script. I was hacking around with the OpenWrt UI under System > Mount Points and finally got that part working.

After that I ran the install script and ended up with the urllib3 issue so I tried your first set of commands above followed by a reboot. That got me up and running! Thanks @b4ng4r4ng!