ihrapsa/OctoWrt #42: I can't start Octoprint

Nico4589, 2023-01-09:
Hi, I did the automatic procedure for installing OctoWrt on my Creality Box 2.0, but without success. When the IP address: 5000 is found on the browser, the page cannot be found. However, I noticed that during the execution of the installation script on the terminal these messages appear:

- Collected errors:  * opkg_install_cmd: Cannot install package gcc.
- warnings.warn( /usr/lib/python3.10/site-packages/pkg_resources/__init__.py:123: PkgResourcesDeprecationWarning: 2013-10-24 is an invalid version and will not be supported in a future release

What caused these errors I used the latest version of OpenWRT following all the steps as described but it doesn't work.
Please help me

ihrapsa, 2023-01-09:
High, sometimes dependencies get updated in pip repository but installation is not tested in OpenWrt. Othertimes OpenWrt needs to be updated to the latest snapshot since the box is not officially supported yet.

ihrapsa, 2023-01-09:
Reinstalling won't fix those issues. I or someone else should find some time to troubleshoot and get to the bottom of the issue. An install log would be helpful to see what's happening.  
If you have any experience with unix and command line or feel brave enough to try some new things you can run the commands from the script one by one and if one of them fails we can look into it

Nico4589, 2023-01-10:
Thanks for the answer, I'll look into what the specific problem is.

cyprusice, 2023-01-11:
I am having the same exact issue here. I have yet to find a solution either. If I do I will keep you posted.

cyprusice, 2023-01-11:
I get this error. 

`Downloading https://files.pythonhosted.org/packages/27/b5/92d404279fd5f4f0a17235211bb0f5ae7a0d9afb7f439086ec247441ed28/regex-2022.10.31.tar.gz#sha256=a3a98921da9a1bf8457aeee6a551948a83601689e5ecdd736894ea9bbec77e83
Best match: regex 2022.10.31
Processing regex-2022.10.31.tar.gz
Writing /tmp/easy_install-0682mvzu/regex-2022.10.31/setup.cfg
Running regex-2022.10.31/setup.py -q bdist_egg --dist-dir /tmp/easy_install-0682mvzu/regex-2022.10.31/egg-dist-tmp-35ok03aa
/usr/lib/python3.10/site-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
error: Setup script exited with error: command 'gcc' failed: No such file or directory`

SamueldaCostaAraujoNunes, 2023-06-20:
> I get this error.
>
> `Downloading [...]

Ola! conseguiu resolver este problema? Estou sofrendo do mesmo...