ihrapsa/OctoWrt #52: Creality Box 2.0 work too ?

freeman5, 2023-12-12:
Hi,

I see it nowhere, can you confirm me its working on WB02 ? 

Thank you =)

freeman5, 2023-12-17:
Someone ,

BJDevelopes, 2024-01-14:
Would like to answer this if you havent figured it out already. I picked up a Creality Box 2.0 recently and tried this and it works but you need to tweak the install script. Follow all steps as instructed until you get to part two, when you get to part 2 you need to tweak the install script to install a newer version of OctoPrint. Heres my tweaked script with instructions, feel free to use.

```
cd /tmp
wget https://raw.githubusercontent.com/BJDevelopes/test/main/2_octoprint_install.sh
chmod +x 2_octoprint_install.sh
./2_octoprint_install.sh
```

gravfu, 2024-01-14:
Here: #54  , I proposed an update. It should be adapted with the version of Python.
 
If you want Octoprint 1.9.3:
```
cd /tmp 
wget https://raw.githubusercontent.com/gravfu/OctoWrt/main/scripts/2_octoprint_install.sh
chmod +x 2_octoprint_install.sh
./2_octoprint_install.sh
```

felipesms, 2024-02-04:
Other error.

ared-workdir/build/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify -fPIC -DFFI_BUILDING=1 -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/OctoPrint/venv/include -I/usr/include/python3.11 -c src/c/_cffi_backend.c -o build/temp.linux-mips-cpython-311/src/c/_cffi_backend.o
            src/c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
               15 | #include <ffi.h>
                  |          ^~~~~~~
            compilation terminated.
            error: command '/usr/bin/gcc' failed with exit code 1
            [end of output]

thanks

Koky05, 2024-02-16:
You need to run before installation of OctoPrint `opkg update && opkg install python3-cffi`. It is mentioned in other issues like 3 times.

gravfu, 2024-02-16:
> Other error.
>
> ared-workdir/build/sdk/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify -fPIC -DFFI_BUILDING=1 -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi [...]

can you try that?

`opkg install python3-cffi`

Koky05, 2024-02-17:
I try but it doesn't help. It ends with same error.

felipesms, 2024-03-02:
https://www.youtube.com/watch?v=I5EKn4aWRdQ

cd /tmp
wget  --no-check-certificate https://github.com/ihrapsa/OctoWrt/raw/main/scripts/1_format_extroot.sh
chmod +x 1_format_extroot.sh
./1_format_extroot.sh

cd /tmp
wget  --no-check-certificate https://raw.githubusercontent.com/BJDevelopes/test/main/2_octoprint_install.sh
chmod +x 2_octoprint_install.sh
./2_octoprint_install.sh

Koky05, 2024-03-02:
I had lots of pain with this repo but in Discord support channel.I found new repo from [shiwajiva](https://github.com/shivajiva101/OctoWrt)